tedsalmon/BlueBus #97: Request: display Siri as caller

Dashdot66, 2021-10-02:
When using Siri the telephone screen is shown but no caller. Ik would be awesome to have Siri as caller displayed.

2001 E38 740i 18:9 screen

tedsalmon, 2021-10-03:
@Dashdot66 

To confirm, you just want it to say "Siri" in the caller ID? This is difficult to implement because I don't know that you're using an iPhone or an Android Device. Maybe I could plop up "Voice Assistant"?

But this begs the question -- don't you know you're speaking to Siri if you summon her? :smile: 

-Ted

Dashdot66, 2021-10-05:
Just Siri is good :)

Actually, the first time I used Siri in combination with Bluebus I thought I had an incoming call. I know it's just a gimmick but if you want to implement it correct you'd have to create a parameter on the device (bt connection) that would allow the user to configure the name of the voice assistent.

dertimaushh, 2021-10-06:
Voice Assistant would be just great.

shaylarihosain, 2022-10-20:
If possible, "Voice Assistant" would be great to have, just to avoid confusion! Thanks so much Ted!