tedsalmon/BlueBus #101: Volume when called

Dashdot66, 2021-11-05:
I don't get called very often when driving but whenever I answer an incoming call I have to crank up the volume real hight to be able to hear the caller. Usually when listening to the radio, not sure what happens when in BT mode. The physical pain is when the call is ended and the radio kicks in on real high volume. Is there any way to adjust the volume of the incoming call?

2001 E38 740i widescreen DSP

tedsalmon, 2021-11-05:
@Dashdot66

What firmware are you on? As of `1.1.17`, all the call volume issues should be resolved for Nav / MID users.

Thanks!
-Ted

Dashdot66, 2021-11-05:
I'm on 1.1.18

tedsalmon, 2021-11-05:
Which beta @Dashdot66 ?

Can you follow this guide: https://github.com/tedsalmon/BlueBus/wiki/Accessing-the-Terminal

Then run this command:
`restore`

You will have to pair your phone back up again, but that will clear out the volume offset. After that, you should be able to raise the volume when in a call and have it go back down once the call ends. Test by calling `611`.

-Ted

tedsalmon, 2022-10-05:
@Dashdot66 

FWIW, 1.3.0 is now available and more feature rich than previous releases, especially in terms of calling.

https://www.bluebus.dev/software

Thanks!
-Ted

closed 2022-10-05 by tedsalmon