tedsalmon/BlueBus #102: E39 MID: TEL Button no longer summons BlueBus Controls

E39NLA, 2021-11-13:
Running 1.1.18 the TEL button as shortcut to BlueBus controls no longer functions, just clears the MID display leaving just the clock. The only way to get to BlueBus is AUDIO then MODE and hope that the CD7-01 is replaced by the BlueBus panel (this is a little inconsistent on my car so I always use the TEL button).

Updated from 1.1.16 where the TEL button functioned correctly.

tedsalmon, 2021-11-14:
@E39NLA 

Hmm. I'm not sure I ever intended for the TEL button to summon the BlueBus, though I could see why it would happen. I'll do some testing later this week.

Thanks!
-Ted

piersholt, 2021-11-15:
@tedsalmon this will be `0x20` in all it's ugly glory.

When selecting "Telephone" from GT menu, the GT emulates the MID telling TEL to display "Top 8".

Ladies and gentlemen, I give you... _BMW_.

E39NLA, 2021-11-28:
Is it possible to downgrade back to 1.1.16 while this is worked on? Being a DSP/MID car it's a lot harder for me to enjoy bluebus if I can't access the correct screen.

tedsalmon, 2021-11-28:
@E39NLA 

Yup! Downgrading is always possible and should be largely issue free.

Thanks!
-Ted