tedsalmon/BlueBus #103: BMW JBIT module - Japan domestic market phone module.

E46er, 2021-11-20:
Hi Ted & others that may be interested.
If you have a Japan spec car with the Jbit phone module, it is possible to easily convert the white 54pin plug used on the JBIT and change it to use the black 54pin ULF/TCU configuration. - so therefore suitable for the adaptor harness for the ULF - Bluebus

see the attached pdf. - (the wire colours mentioned are from a E39, so E46 are slightly different, but pin positions are the same.)

[JBIT_to_ULF_wiring.pdf](https://github.com/tedsalmon/BlueBus/files/7574221/JBIT_to_ULF_wiring.pdf)

dertimaushh, 2021-11-20:
Can confirm this is working :-)

tedsalmon, 2022-10-08:
Thanks again for this! I'm closing this as there's no further action required :)

closed 2022-10-08 by tedsalmon