tedsalmon/BlueBus #104: Oil Temp on 1.1.8 - on 2000 E38 - M62B35TU

dertimaushh, 2021-11-21:
Hi Ted, 

Just did a highspeed run on the Autobahn.  Water and Ambient Temp was displayed correctly. 

But Oil readings were rubbish. 177 degrees celsius (!), next moment 235. Switched back and forth and then it showed 50-60 degrees celsius while doing >200km/h over 30km. Back in town it showed 205. 

Any idea for this behavior? Or is this calculated value just pointless? 

Besides that - 1.1.18 works great.  

Back in love with my E38. 

Best regards,
Tim

tedsalmon, 2021-11-21:
@dertimaushh

Yeah, you're not the only one to mention issues with the value. Off the hop, I can see that I forgot support for the LCMIV -- can you try this revision of the firmware? https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_1_18_1.hex

Thanks!
-Ted

dertimaushh, 2021-11-22:
Don't think this will solve it for me as I am not running an LCM4. 

The sensor itself was replaced just 6 months ago. 

Will do some reading with Inpa to check.

BR,
Tim

piersholt, 2021-11-22:
@dertimaushh can I confirm which light module you're running in your E38?

@tedsalmon do you believe this might just be the calculation? If it's to do with LM variants or cluster type, we might be at cross purposes.

CaptnO, 2021-11-22:
I am running a LCM IIIb in my E38 and I seem to get valid values and wrong ones as well. There might be an issue with the calculation of the value, since as far as I remember it is calculated through two or three different sources.

dertimaushh, 2021-11-22:
Looking at my production date - it's an LCM 3b aswell.

CaptnO, 2021-11-22:
I am not sure about that. The E38 never came with a LCM 3b when stock. That would explain why I had to add the C18 value to get it properly coded. The only difference to the LCM 3 I know is the automatic headlights feature. What does INPA say? It should tell you the exact module.

dertimaushh, 2021-11-22:
Yes, ok will check the readings and Version with INPA.

tedsalmon, 2021-11-22:
@piersholt @dertimaushh @CaptnO 

I did fail to mention that I updated the calculation from what I ripped out of the guts of `IBus App` to what RichardP of the Intravee shared with me a couple years ago. I wish I could test, but I only have E46s :/

```
+            uint16_t offset = 310;
+            if (ibus->lmVariant == IBUS_LM_LCM_IV) {
+                offset = 510;
+            }
             // Oil Temp calculation
-            float rawTemperature = (pkt[23] * 0.01275) + (pkt[24] * 0.000050);
-            unsigned char oilTemperature = 1.0 * 67.2529 * log(rawTemperature) + 310.0;
+            float rawTemperature = pkt[23] * 0.00005 + pkt[24] * 0.01275;
+            unsigned char oilTemperature = 67 * log(rawTemperature) + offset;
```

piersholt, 2021-11-22:
@dertimaushh @CaptnO 

I'd recommend using INPA to determine the light module variant. The reason being, parameters such as coding index, hardware number etc aren't a reliable means of identifying the variant. I say this having reverse engineered BMW's [variant identification system](https://github.com/piersholt/ediabas-sgbd/blob/master/D_00D0.B2G) (which INPA uses).

The variant will be listed under _Info (F1)._

[image]

The E38 _should_ have an early light module variant as the later light module variants included the CCM which is discrete in the E38.

CaptnO, 2021-11-24:
@piersholt I just mentioned having a LCM 3b, since I replaced mine to retrofit the automatic headlights feature :).
I had a LM38 before, which is in fact a LCM 3.

I just want to add, that I realized that the oil temperature is set on 104°C on startup, which seems like a fictional value (or the last plausible when the car was running before). Once the engine reaches it's operating temperature, it seems to work (at least for me). There are times by the way, when the Bluebus doesn't show any value for the oil temperature. I wonder why what that is.