tedsalmon/BlueBus #105: Can't recover from BT connection timeout

Ron199999, 2021-11-26:
If not connecting to the module within 10-15 minutes from its power up, connection will time out and become not possible.
Noticed that now that i've started to let my car idle in the driveway before i start driving it.
I will start it (Bluebus will turn on) but i will not have my phone near to connect but after 10-15 minutes.
At this time - Bluebus is visible in the phone BT menu, but the BT module would return "Unable to connect" to a connection attempt not will it connect automatically.
this is resolved if i go to the trunk and power cycle the bluebus module - it'll connect automatically to my phone.

Details steps:
1. Pair phone with BB module
2. power down BB
3. power up BB without paired phone present
4. wait 10-15 min (dont have specific amount of time, i assume it's a timeout / max num of retries )
5. get paired phone present - will not connect
6. try to connect manually through BT menu - will not connect (Unable to connect error on phone)

Hope this helps.

tedsalmon, 2021-11-28:
@Ron199999

This is a great report! I have no idea if this has ever worked, though it definitely should.

I know there's a timeout built in for re-connections, but scans should continue regardless.

I'll look into this!

-Ted

Ron199999, 2021-11-28:
Back in 2003 i did some QA for a small start up, i could relate to the frustration of getting feedback without specific step to reproduce a scenario (also note that the first post was after a long drive home w/o music and def after multiple glasses of wine ;)

Let me know if there's anything else i can help with, i assume car details are not extremely important, but this happened on a 2000 M5 with CD43, DSP amp, 16:9 front screen and Nav Mk4, no phone module installed.