tedsalmon/BlueBus #107: E83 X3 - Conflict with factory satellite radio

ryannoblett, 2021-12-25:
I have a 2007 X3 3.0si with factory satellite radio (2006 build date). I installed the bluebus and disconnected the TCU, but I'm unable to locate a satellite radio module - it's not in the trunk area where I expected it. The bluebus sort of works, I can receive phone calls, and if I mess with the buttons for a bit, I can get into the menu (with glitches), but can't play music and only hear the satellite radio demo channel. Also, the comfort blinkers don't work, etc.

My assumption is that I need to remove or disable the satellite radio somehow. Is this correct?

If so, does anyone know where the satellite radio module is on this model? I've been unsuccessful in locating it. It's not in the driver side trunk area with the TCU and navigation DVD drive like it is on most other models. I also checked under the cargo floor area.

Thanks for your assistance.

tedsalmon, 2021-12-25:
@ryannoblett

Can you describe the menu "glitches"? This vehicle is equipped with Navigation, correct?

>My assumption is that I need to remove or disable the satellite radio somehow. Is this correct?
Not entirely. You can if you don't want it, or you could use the slave cable: https://www.ecstuning.com/b-genuine-bmw-parts/sirius-slave-cable/84110150526/?

Of course, you'd have to find the Sirius module first. I do believe it should be in the rear driver side electronics bay, along with the Navi and other modules.

Did you install the BlueBus in that area as well? Any pictures of how it's installed? Typically these cars had the CD Changer installed in the center console, which is why I'm asking.

It may also be easier to reach me by email: support@bluebus.dev

-Ted

ryannoblett, 2021-12-28:
Yes, it's got navigation. When I say glitches, sometimes the text in the bluebus menus get overwritten by other text I assume is from the satellite radio module. I wasn't able to capture that well, so I hope the description is good enough.

I did install the bluebus in the driver's side trunk area with the navigation drive and TCU. I'm attaching some pictures, and I'll link to a video trying to use the bluebus, but hearing the satellite radio instead.

Video: https://pics.deadface.org/phone/x3/bluebus/VID_20211227_194148.mp4

[image]
[image]
[image]
[image]

Thanks again for your help.

E39NLA, 2023-03-24:
Following this thread as I also have an E38 X3.

I think you may have the Bluebus connected wrong. According to TIS, an X3 with Sat, Nav, and DSP should use the 6-pin CD-SDARS connector. As Ted mentioned, the CD changer on these cars lived in the center console box, not the rear equipment compartment. You might find that connector there. See attached.
[image]

ryannoblett, 2023-03-24:
Thanks for sharing this info, turns out it's exactly what I needed. I pulled the tray out of my center console to find both an aftermarket Sirius satellite module and a Dice Mediabridge!
[image]
[image]
I'll get this stuff out and see what it takes to get the bluebus going.

ryannoblett, 2023-03-24:
So, partial success - the bluebus plays music perfectly when plugged into the CD changer connectors in the center console.
[image]
[image]
However, there is no way to connect it to the TCU adapter plug, which is in the trunk area. I think I will have to make an extension cable and run it back there to get Bluetooth calls working again. Unfortunately, even with the Sirius and dice Mediabridge disconnected, it still does not work correctly using the connectors I'm the trunk area.

ryannoblett, 2023-03-25:
Thanks Ted, switching that cable made it work in the rear and now it's fully functional.

closed 2023-03-25 by ryannoblett