tedsalmon/BlueBus #108: while installing latest firmware

socoolbreeze, 2022-01-04:
Hi @tedsalmon,

I don't know if this is the right place for all the feedback but let me know.

Saw the latest update 1.1.18. I’m very curious to try it out tomorrow. It’s late here in Netherlands.

A lot of BT commands don't work for me (not firmware related, on older same outcome)

See picture like BT CONFIG. I think I should see a lot more?
Also setting a pincode doesn’ t work. It replies with OK. But when adding it with any device it's open for pairing without pin(key).
The feedback is also a simple ok with no standard position as you can see.
[image]

The get commands works like a charm. I like the option when you type a command without a paramater, that you get the setting status. This would be great to quick check/verify bt name, bt pincode etc. or make a setting-status report when connected to computer.

Also wondering the bootfailures of the BC 127 related to the pin setting not working.

gr Ben van Maanen

tedsalmon, 2022-01-05:
@socoolbreeze

Did you take that screenshot with the unit in the car or inside? How cold is it? If you had just brought the unit inside from the cold, that is important to note as well.

It does appear that the BT module has failed to boot, which makes sense given your Boot Failure counter. Do you have connectivity issues?

>This would be great to quick check/verify bt name, bt pincode etc. or make a setting-status report when connected to computer.

This is difficult to do because these things are part of the output of `bt config`, so I'd have to parse them out. That's why `bt config` exists in the first place, but of your BT module failed to boot so it wasn't responding.

I can of course fix the issue with it not booting, but I would need you to mail it to me.

-Ted

socoolbreeze, 2022-01-07:
Hi Ted,

It had the bluebus around an hour inside the house.
Around 5,5 Celsius outside. The latest firmware didn't work for me. Screen was almost empty (bluetooth was replaced with half undertermined word) and quickly flashed it back to 1.1.6. 
Same screen but now toggled sources and the display button and hit the oke nob and worked.

After vacantion i'll try 118 again with more time.

BT connection was always oke. Stable. Only BT trouble was when adding a second phone. After few attemps it worked.
Only when switching to other phone sometimes the name of other phone was scrambled.

With fw 118 attempt it started with one hickup but while riding to garage it was solid in playing music.

The br config is hopefully working better when BT not restarting that often.

After vacation and testing FW 118 i'll send you an update and we'll see if I have to mail it to you. Or via remote desktop (win or mac)...if its software only.

grtz Ben van Maanen

nabucho, 2022-02-21:
@socoolbreeze - I noticed that with 1.1.18 the BT CONFIG actually writes the data only when you have logging for BT enabled. Same for BT VERSION command.

Can you test:

# set log bt on
# bt config
# bt version

And let us know if you get the data from BT?

socoolbreeze, 2022-02-27:
Thanks for the tip @nabucho . It works and I see the BT name and pin etc. etc. thumbs up

I've been able to install 1.1.18 firmware a few weeks ago and somehow it worked.

Still just a few glitches in 1.1.18 on my e39

-Language Update- Diffferent value
Saw a few new options in UI & comfort (jeeej) and I want to update the dutch translation.
In same google doc as before?!

[image]
In settings/sound the volume value is different in dutch.
Also with the new vol rev. function you see it say's AAN =On in English but in english menu it is set to off.

-Vol Rev.
I like the music dimming when going in reverse gear but I think the PDC now works les soon as before.
Usually i'm not to fast and I recently bumped a car. Testing it now switched off and have to test a few times to know for sh

I like the temps on the dash...have a character thing, is it because of eu screens options?!
It also disappears sometimes. 
[image]

The rest works really good so thanks for the extra features.
Love the parking lights 😎 could it be set to longer (i'm a slow walker) like the blinker a few options?!

Many thanks already for your comments.
greetz from the Netherlands

nabucho, 2022-02-28:
@socoolbreeze , please feel free to look at the localizations here ( not sure about previous doc, but this is one I just made for you ).
https://docs.google.com/spreadsheets/d/1--IeTjznZmvI2CP8ASKyac1zsPa1fPcR1-Ehs1QBhGI/edit?usp=sharing
The wrong volume display is also caused by error in this localizations, so please update the texts and will fix in one go.
Same as for the incorrect vol.rev translation.

Thanks for the Temperature on Dashboard screenshot, helps a lot. Can you also try to make one with "split-screen" setup?
What you mean by "have a character thing, is it because of eu screens options?!" - is it the + sign at the end? Normally it would be followed by outdoor temperature, + is part of it. It just seem to not fit on screen. Or is there anything else you have in mind?

For others, I will let @tedsalmon reply.

nabucho, 2022-03-01:
@socoolbreeze can you flash your firmware and test the translations and UI of temperatures, please? In this build there are a couple of more stability updates, so if you can report on any other issues. Photos of temp display, eg. wrapped, missing text, helps. (unzip the file to get the hex to flash) 
[application.production.hex.zip](https://github.com/tedsalmon/BlueBus/files/8165146/application.production.hex.zip)

tedsalmon, 2022-03-02:
> -Vol Rev. I like the music dimming when going in reverse gear but I think the PDC now works les soon as before. Usually i'm not to fast and I recently bumped a car. Testing it now switched off and have to test a few times to know for sh

Sorry, are you saying that PDC is acting differently since you enabled volume dimming in reverse? That shouldn't be possible as I only passively listen for the PDC messages (along with the transmission gear state) and don't issue any commands on the I-Bus.
 
> -Steering wheel controls Had 2 times that i want to skip number with steering controls but I change radiochannels as prompted on IKE but the bluetooth music keeps streaming.
So the IKE showed you were on the radio but Bluetooth was playing? Did it eventually pop to the radio on its own? Can I assume you are using "DSP: Digital" to force the use of the coaxial interface? If yes, I can explain, but I'll wait for your input. This is likely related to a long standing bug.
 
> The rest works really good so thanks for the extra features. Love the parking lights sunglasses could it be set to longer (i'm a slow walker) like the blinker a few options?!
The fact that the AEs stay on after you get out of the car is a bug not a feature, lol. That "feature" is still to come!

> @socoolbreeze can you flash your firmware and test the translations and UI of temperatures, please? In this build there are a couple of more stability updates, so if you can report on any other [...]

Thanks so much, Ivan! Do you have this in a branch somewhere so I can take a peek? :)

-Ted

socoolbreeze, 2022-03-04:
You're right @tedsalmon The pdc beeps stays consistently late so sorry for doubting the bluebus.
The oil temp worked perfect in 1.1.17beta2

IKE display showed radio but bluetooth was still playing music. But not controllable anymore. Yes I remember the digital setting. I'll check the analog option.

Partly non bluebus related problem.
Sometimes I use He Siri command and the voice goes over bluebus but it says that it cannot connect and there is no homepod. I'm using latest ios 15.3.1 but I have it also in rental car with bluetooth connection. Anyone else same problem?

I'll hope to check the latest language update soon @nabucho My e39 530i is currently in repair because of transmission problems.

Thanks for the updates 🙏

socoolbreeze, 2022-03-20:
Hi @nabucho
I got my car back yesterday with replaced gearbox..just quickly checked the lastest update with language. I corrected my mistakes in the dutch translation doc. Dash still looks like this.
[image]

greetz Ben

socoolbreeze, 2022-06-29:
> @socoolbreeze , please feel free to look at the localizations here ( not sure about previous doc, but this is one I just made for you ). [...]

I changed the file with language. I hope you can update it into next FW.

tedsalmon, 2022-06-30:
I'll try to take care of this early next week :)

socoolbreeze, 2022-06-30:
> I'll try to take care of this early next week :)

🙏 thanks!

socoolbreeze, 2022-10-08:
Hi @tedsalmon I was trying to install latest firmware but could not connect
I used same cables, reboot etc. but no response.
I can connect via terminal with the bluebus but the firmware tool doesn't see it.

This is the terminal screen
[image]

tedsalmon, 2022-10-08:
@socoolbreeze 

Based on your screenshot, something has the serial port open. Can you check if `screen` is still open?

-Ted

socoolbreeze, 2022-10-08:
Yes that was it probably! I was to quick and unexperienced.
It feels like unpacking again jeeej.

Thanks for this update, i'm curious!

closed 2026-02-25 by tedsalmon