tedsalmon/BlueBus #111: BlueBus Not Working, Undetectable by Computer

OneSwellFoop1, 2022-03-05:
I didn't drive my M3 for a couple of months as it's a weekend/summer vehicle and it was cold.  I experienced battery drain (possibly attributable to the BlueBus, but it's a 17 year old German car, so who knows) and had to remove the battery and charge it.  Upon reinstallation of the battery, my BlueBus did not work. 

I attempted to connect it to my computer and neither of my computers will detect it as a USB device, despite trying two different machines and multiple USB cables.  Power light on the unit comes on.  It got quite cold in the trunk of my car during the winter, but never below mid 20's F, and there was no water intrusion or other environmental thing that might affect it.  Device was room temperature and had been for hours when I tried to connect it to a computer via USB.

tedsalmon, 2022-03-05:
Hello,

For what it’s worth, email is a better support medium for hardware issues :)

If the power light is coming on, then the unit isn’t dead. It’s likely you’re missing the drivers — can you see: https://github.com/tedsalmon/BlueBus/wiki/Accessing-the-Terminal#troubleshooting

When you plug it into the car, does the green light come on?

Did you try unplugging it and plugging it back in before fully removing it?

-Ted

tedsalmon, 2022-10-08:
Closing this due to lack of response. If it's still an issue, shoot me an email -- support@bluebus.dev

closed 2022-10-08 by tedsalmon