tedsalmon/BlueBus #112: Setup bt pairing pin

morrow95, 2022-03-24:
Any chance for some instructions on setting up a pin for bluetooth pairing?  I see it mentioned in a few of the videos, but can't find any specifics anywhere.  Using putty for terminal connection is not a problem and I am very comfortable using it.

morrow95, 2022-03-24:
Guess all I needed to do was connect with terminal and look at the commands - duh.  With that said, I can't seem to get the pin setup.

bt pin 1230
bt reboot
bt pair

Then I see 'BlueBus' as a device on my iphone and connect.  It never asks for the pin or anything and stays 'not connected' according to my phone.  Will this only work when connected to the car?

socoolbreeze, 2022-03-24:
It doesn't work for me either. But since the BT only shows when you are in pair mode it's not really a big deal to not have a pin. But yeah it should work.

tedsalmon, 2022-10-08:
@morrow95,

I got around to trying this and was able to replicate the issue. Unfortunately, this is a bug with Melody, the BC127 firmware. I verified the PIN was correctly set in the device configuration but was still not prompted for it :(. I'm going to have to close this as a known issue because Sierra Wireless isn't looking to fix any issues with the BC127 at this point, unfortunately.

Sorry!
-Ted

closed 2022-10-08 by tedsalmon

socoolbreeze, 2022-10-08:
Ok thanks! Maybe remove the option then when connecting with Bluebus depending on hardware.

tedsalmon, 2022-10-08:
> Ok thanks! Maybe remove the option then when connecting with Bluebus depending on hardware.

Yup, that's done on working copy of the code :)

morrow95, 2022-10-10:
Geez.  That seems like a pretty important/standard feature of bluetooth in general for them to not care about updating the firmware.  Is this in both bluebus version or just the v1?