tedsalmon/BlueBus #114: Bluetooth not showing up, 2000 e38

tbecker82, 2022-05-09:
Hi,

I tried to install my Bluebus today, removed my CD player and hooked everything up, green light came on, but when I tried to pair my phone I have no option Bluetooth in my radio.  When I go through the modes CD still comes up but it says disk 7.  

Here is an image of what shows up.

https://i.postimg.cc/P5X05zq2/IMG-20220509-160235.jpg

Thanks.

tedsalmon, 2022-05-10:
@tbecker82

Is this a video module or a full on Navi? My first suggestion would be to install the newest software: https://d3uwtdqahrxbn.cloudfront.net/BlueBus/firmware/bluebus_1_1_19_rc_11.hex

If that fails, then follow this guide: https://github.com/tedsalmon/BlueBus/wiki/Accessing-the-Terminal

Then run this command:
```
set ui 2
```

-Ted

tbecker82, 2022-05-10:
Hi Ted,

It's a full nav system.  Ill give your 2 suggestions a try.

Thanks

piersholt, 2022-05-10:
@tbecker82 the early navigation systems are a bit of a mess.

To be 100% sure, can you confirm the version ID in the upper right hand corner of the _SET_ screen?

[image]

tbecker82, 2022-05-10:
Here is my version ID
[image]

tbecker82, 2022-05-10:
Ted,

So I just updated the firmware and was able to get my phone connected.  Thanks for the help.
On another topic, are you selling the phone adapters separately now?

Thanks

Tony

piersholt, 2022-05-11:
Glad you got it working @tbecker82!

Wow, that looks like MK2 nav still running software it came from the factory with!

tbecker82, 2022-05-11:
@piersholt 
I wouldnt be surprised if it is.  The car was a one owner lady who had passed away and her 
kids were selling it.  43k original miles :).

closed 2022-05-11 by tbecker82