tedsalmon/BlueBus #116: E39 - MID Display Fighting with TCU

tedlee1688, 2022-06-28:
After disconnecting the battery and the Bluebus to do some work on the car, the BlueBus is now conflicting with the bluetooth phone TCU that I have in the car.  When I pull up the BlueBus on the MID (non-nav), it keeps switching back and forth between the TCU and the Bluebus, and commands to the BlueBus also get sent to the TCU.  Any ideas?  Should I try switching iBus connectors in the trunk?

tedsalmon, 2022-06-29:
> After disconnecting the battery and the Bluebus to do some work on the car, the BlueBus is now conflicting with the bluetooth phone TCU that I have in the car. When I pull up the BlueBus on the MID [...]

The Bluetooth TCU and the BlueBus are not compatible with each other on the MID UI. The reason is that the BlueBus hijacks the Telephone UI to provide a stable interface. Let me take some time to re-evaluate the feasibility of stabilizing the UI without this "hack".

-Ted

tedlee1688, 2022-06-29:
No worries I can live with it for now. Weird that it's doing it now though
as I don't remember having this issue before I did the battery pull last
month

sleuth255, 2022-11-19:
It sure would be cool if a hack could be found to place bluebus settings where they belong: on the main settings menu.  That would eliminate this condition and make BlueBus integration more standardized IMO.  There are definitely optional components that do this once they are installed: TV and the bluetooth ULF leap to mind here.

rickybobby80, 2022-11-19:
Sirius Radio is another

Rick