tedsalmon/BlueBus #117: Feature Request - Status? - SMS/GPS integration

60SecTheBaptist, 2022-08-11:
is there a plan for SMS(cellular) and GPS integration so that BlueBus could indicate the vehicle's various states and notify the owner? 

- vehicle on / off
- location
- doors locked unlocked 
- battery charge (for when the veh is stored)
- Tilt alarm (towed)

*** obviously, this requires a power source other than the veh battery (like a power brick) but that could be integrated by the owners,   I'm just curious about the SMS/GPS

ShonkyCH, 2022-08-12:
Where is the cellular service coming from? You don't leave your phone in the car - needed for of those functions. Any BMW Assist/SOS is highly unlikely to still be function even if it were capable of sending custom messages etc.

Also the nav GPS in a standard car shuts itself down so position isn't available unless vehicle is woken up. Not great for battery life.

A proper GPS vehicle tracker will get you some of that functionality - they include both GPS receivers and modem modules of their own.

Sorry it just doesn't really work that way.

piersholt, 2022-08-12:
In short, no.

BMWs of this era do have telematics capabilities, although, they're rather rudimentary and require optional equipment.

However, as @ShonkyCH points out, the BlueBus, when equipped, replaces a core component of the factory telematics system- the telephone!

Is the functionality you're discussing _technically plausible_? Yes, but it's highly impractical. I'd agree with @ShonkyCH, and say that a dedicated vehicle tracker is far more cost effective.

closed 2022-08-12 by piersholt

60SecTheBaptist, 2022-08-12:
Ok, thanks for replying