tedsalmon/BlueBus #119: Beep when coolant hits thermostat opening temp

bcagnoni, 2022-08-15:
The nightmare with our beloved engines is the thermostat failing to open.

I found personally great the coolant/oil temp feature. A nice enhancement would be a "beep" similar to the one at 3 Celsius, at the thermostat opening temp. In my case with a M54B30 engine it 97 C but could be different on other engines.

Beside the obvious benefit of making sure the thermostat opens and you can consider being safe for that run (very unlikely the thermostat fails closing when it's already open and the engine is in operating temperature) there will be a pre-warning in case the engine starts overheating for any reason.

@tedsalmon I guess the easiest way of implementing this it's just set the temp manually chosing from a pre-populated list. I'm happy to start collecting all temps for the BB supported models.

ShonkyCH, 2022-08-16:
How does a beep at the thermostat temperature tell you it's opening/working?

Otherwise it's an "everything's ok alarm"? Alerts should generally be for abnormal operation IMO.

tedsalmon, 2022-08-17:
@bcagnoni 

It is only possible to request that the IKE alarm the user when the IKE in question is the "High OBC" version. Otherwise, lower cluster can't really be made to do anything other than display distance to the next turn (for navigation).

I also agree with @ShonkyCH that alarms should only be implemented when issuing caution to the end user. Otherwise, it turns into "the boy who cried wolf" -- you might get so used to the beep that you stop noticing it.

So while I can implement this, I don't agree with the proposed implementation method and it would only work for a small subset of vehicles (higher trim E38s, E39s and E53s).

Thoughts?

-Ted

piersholt, 2022-08-17:
@ShonkyCH raises a nuanced but I think important point regarding alerts. When you introduce 'informational' alert, you begin to mix metaphors. The purpose of BMW's Check Control is to "_inform the driver [...] regarding failure of important functions as well as notes and warnings._", which contrasts with the proposed enhancement, the purpose of which is ostensibly say the opposite- "nothing is wrong".

It's also worth noting that, at best, we're _inferring_ the condition of the thermostat based on the coolant temperature. However, reaching a target coolant temperature isn't necessarily indicative of thermostat condition.

For example, a thermostat I removed from an M52 had a ‘soft failure’, in which it opened as expected, however it closed very slowly. The only indication of failure was when driving at highway speeds (high airflow, low load).

As another example, I've experienced a thermostat failing in an open position, however, not open enough to prevent the coolant from reaching the target temperature. It was actually the notable reduction in fuel efficiency from cold start which tipped me off!

Lastly, the coolant temperature is likely to reach this point any given number of times during operation. This is why the cluster’s coolant temperature gauge has a non-linear scale. The ’normal operating temperature’ (when needle is centred), represents (in the case of the IKE), a temperature between 75°C- 110°C!

bcagnoni, 2022-08-17:
I understand the points you made and I partially agree. I missed to specify one aspect about having an alert at the thermostat opening temperature: it brings attention at the temperature at that specific moment in time and in the next minute or so. If the coolant temp goes down it's all good. If it keeps going up you have a problem. Make sense?

In my experience it goes up to 97°C (I've seen 99 just once) after that when it gets around 95°C the fan kicks in and the in matter of 20'/30' the temperature is at 92°C. 
I don't know what's the behavior in the models with the fan clutch, but I don't thinks makes big difference in this specific case. 

If the thermostat gets stuck closed, getting to 110 it's really quick. The ’normal operating temperature’ in my opinion it's a cosmetic thing, just for not having the gauge moving all the time. 
As extremes having the engine operating long time at 76°C or 109°C is non healthy but according to the logic behind it, it's all fine.

@tedsalmon If the applicability is restricted to only a few models I understand it could not be so interesting to implement.

Just my thoughts.

piersholt, 2022-08-19:
> If the thermostat gets stuck closed, getting to 110 it's really quick.

Okay, this is a good example of what I mean by thermostat condition being _inferred_ based on coolant temperature, and so in turn, why a caution/warning model is optimal.

The envelope for 'normal operating temperature' is large owing to the variety of environments in which these vehicles are required to operate. Both this temperature- 110°C, and the rate at which the coolant reaches this temperature- 'really quick', is dependent on said operating environment. That is to say, while it mightn't be for you, this temperature-  110°C, is a normal operating temperature in _some_ conditions.

This is down to the efficiency of the cooling system which varies depending on a number of parameters. For example, the efficiency of the cooling system of a vehicle operating at sea level on a cold dry day will be higher than if that same vehicle was operating at altitude on a hot humid day. This is of course, further compounded by the manner in which the vehicle is driven.

My longwinded point is, the mechanism by which you're wanting to gauge the condition of the thermostat is essentially arbitrary (i.e. it's a guess). This is in contrast to the coolant temperature alert which is based on a directly observable property of the coolant itself, and of which there is a known threshold.

.

This, naturally, brings me to another point. It's not the thermostat that matters, but what it's designed to regulate- the coolant temperature, and until such a time the coolant temperature exceeds known thresholds, the thermostat condition- with respect to the operation of the vehicle, doesn't matter. Sub-optimal, yes, but fatal, no.

This brings me back to the point in my first post about 'informational' alerts and mixed metaphors...

bcagnoni, 2022-08-21:
@piersholt normal operating temp for these engines is 95~97. Also consider G11 coolant 1:1 with distilled water as per repair manual has a boiling point at ~107C. You never want water to boil in your engine.

My all point is about to spot the worst case scenario of a stuck close thermostat where the temperature can raise way beyond the 110C in matter of few minutes.

In practical terms the difference between stopping the car when you see the temperature constantly raising before it reaches 103-105C and waiting to see the gauge moving to the right and the red light on it could be a lot of money.

This is just common sense and using a reliable information to infer that there is an anomaly.

bcagnoni, 2022-08-21:
> So you're then talking about a second beep? Or you wait for the first beep and then manually watch coolant temperature guage which is signifcantly buffered in non-M cars (but less so in M cars) as mentioned so doesn't really tell you much.
> 
@ShonkyCH just one beep at a temp that can be configured under Settings. M cars don't need that I agree.

piersholt, 2022-08-23:
> Also consider G11 coolant 1:1 with distilled water as per repair manual has a boiling point at ~107C. You never want water to boil in your engine.

You've unwittingly made my point for me 😋

Coolant- in your case 'G11', diluted 50/50, with a boiling point of 107°C... _won't boil at 107°C_.

Madness, surely? Not quite.

Take another look at the data sheet for your coolant, specifically at the testing method listed for boiling point. I'm guessing it's [ASTM D 1120](https://www.astm.org/d1120-17.html).

> This test method establishes the standard procedures for the determination of the equilibrium boiling point of engine coolants. The equilibrium boiling point indicates the temperature at which the sample will start to boil in a cooling system under equilibrium conditions **at atmospheric pressure**.

However, our cooling systems do not operate at atmospheric pressure. The system is _pressurised_.

If you take a look at your radiator cap, you should see a number- likely 200, which is the pressure the cap (a regulator) is rated to, in this case 200 kPa, or about 29 psi.

At the maximum pressure of 200 kPa, the boiling point is not ~107°C, but closer to ~137°C.

.

This demonstrates why the proposed mechanism is arbitrary. Your mental model for 'correct' thermostat operation is based on an incomplete understanding of the cooling system. You may as well flip a coin! 😇