tedsalmon/BlueBus #122: FW 1.2.2 - BlueBus doesn't autoplay

steveooooooooo, 2022-09-03:
Hi Ted,
When running 1.2.2 on the new hardware, I'm not able to have the bluebus autoplay and resume paused music when I restart the car. Is there a setting I should adjust to change this? The car is a 2002 E39 w/ MID w/o DSP.
Thanks,
Steve

tedsalmon, 2022-09-03:
> Hi Ted, When running 1.2.2 on the new hardware, I'm not able to have the bluebus autoplay and resume paused music when I restart the car. Is there a setting I should adjust to change this? The car is a 2002 E39 w/ MID w/o DSP. Thanks, Steve

Under "Settings" there's a toggle for Autoplay, and it is off by default. If that doesn't do the trick, can you see if it autoplays when you go to FM and then return to Bluetooth mode?

steveooooooooo, 2022-09-05:
Thanks, Ted. I'm not seeing a setting for Autoplay on the MID or in Putty. On the MID, I have:

- Hands free 
- Metadata 
- Lower volume on reverse 
- Car
- Comfort blinks 
- Comfort lock
- Comfort unlock 
- Dsp
- Clear pairings 

Here is what I have in Putty:
- [image]

Also - I may have used to Autoplay to discuss two separate issues. Both of the following are happening: 

- When I am listening to the BlueBus and then turn off the car and then come back after a bit, the source reverts to FM.
- When I manually return to the BlueBus source (whether the car is on or off), the audio does not automatically resume.

Let me know of your thoughts on this. No rush with today being a holiday.
Thanks!

tedsalmon, 2022-09-24:
@steveooooooooo,

Autoplay is off by default, and I just realized that I broke it at some point and made it inaccessible on the MID :(

I just fixed the code, but I need to validate the change. For now, this should get it working:

`set byte 0x40 0x01`

I also recommend you update to this firmware version: https://d3uwtdqahrxbn.cloudfront.net/BlueBus/firmware/bluebus_1_2_3_f12.hex

Thanks!
-Ted

steveooooooooo, 2022-09-25:
Thanks, Ted. All sounds good. I have the new firmware installed and used the set byte command. I tested a bit and autoplay seems to work - I'll continue to try it out over the next couple of days.

The other piece that hasn't changed is that the source still consistently reverts to FM when I turn off the car and return. I'm thinking that modifying the PWROFF or IGN settings might help with this. Do you have any insight on which one / what setting to use or any that I should specifically avoid because of battery drain? I'll work on testing this as well.
Thanks as always,
Steve

steveooooooooo, 2022-10-02:
Hi Ted - Good news. In addition to the set byte update, setting PWROFF to off has brought the bluebus to functioning as I'd hoped. Upon startup, the MID returns to the bluebus (if it was in use when I left the car). The audio also resumes via autoplay as expected.
Thanks again,
Steve

closed 2022-10-02 by steveooooooooo