tedsalmon/BlueBus #123: FW 1.2.2 - Where do I get it?

kaikoehler99, 2022-09-18:
As the title says: where do I get the FW 1.2.2.

Problem is - for sure it was my fault - I got the BlueBus with the hardware rev. 2 and I installed the latest firmware available on the BB website (as I usally do with new hardware: first get the updates from their website).

Now nothing really works anymore - there was no problem flashing the firmware on to the BB but now there is no way to get the bluetooth connected to my phone again, also the BB doesn´t appear as a telephone on my E46 anymore.

My assumption is that the firmware listed on the website is for the 1st hardware rev. is that right?

piersholt, 2022-09-19:
I've been out of the loop on firmware for a few months, but I believe it's 1.1.18 available via the BlueBus website.

Ted discusses 1.2.2 on [M5Board](https://www.m5board.com/threads/fully-integrated-bluetooth-solution.603108/page-52#post-7672378). Post number 1024 includes links to firmware.

I believe 1.2.2 is still an RC build, but IIRC 1.1.18 doesn't yet have support for new hardware.

tedsalmon, 2022-09-24:
@kaikoehler99

Sorry for the late reply -- this is currently the newest firmware: https://d3uwtdqahrxbn.cloudfront.net/BlueBus/firmware/bluebus_1_2_3_f12.hex

As @piersholt mentioned, anything prior to 1.2.0 is NOT compatible with the second hardware revision. 1.2.4 should be out next week and available for all BlueBus users regardless of hardware revision.

Thanks!
-Ted

closed 2022-09-24 by tedsalmon