tedsalmon/BlueBus #124: Question: Possible to enable metadata with *no scrolling*

8one6, 2022-10-05:
I understand there is only a fixed amount of real estate to display metadata, and the docs seem to indicate there are two scroll modes (chunk/party) to move through the metadata on the limited screen real estate.  Is there an option to display as much as will fit and then *not scroll at all*?  I would appreciate seeing some of the song info, but would prefer to avoid the distraction of a constantly changing display.  (Changes when moving to the next song are fine.)   I know I can switch to displaying the word "bluetooth" statically if I want to avoid distraction, but was hoping for the middle ground.

tedsalmon, 2022-10-05:
@8one6

Isn't that a little useless though? For example, the CD53 screen only has 11 usable characters which isn't enough to fit even one song name :/.

8one6, 2022-10-05:
First, thank you so much for this wonderful device.  I'm waiting for my ECS shipment and am really excited!

Second: I'm coming from a "mediabridge" and the compromise I described above is one offered by that device.  I don't know how things will work with the BlueBus, but on the mediabridge, there was a chirp of static played over the speakers each time the device tried to update the display.  So you either had to disable scroll (to spare your ears) or deal with once-a-second small blips in the audio.  

I went with the former choice (to avoid going totally insane).  Once I got used to it, it was fine (better to have some song info than none), and I actually preferred the lack of distraction.  Put differently: I'd much rather have 70% of a song title displayed than "bluetooth" at all times.

tedsalmon, 2022-10-06:
@8one6,

Thanks for the support!

The chirp you're hearing is the screen updates -- you're hearing data travel across the car, which while annoying, is pretty neat. Now, I'm not going to say this noise doesn't exist with the BlueBus but it basically doesn't. Lots of time was spent overcoming the issue and I finally settled it by dropping the audio load resistor to 100ohms. If you crank the volume up quite a lot and then silence playback, you'll still hear that noise, but never under normal playback conditions.

I will look into adding this as an option. Maybe I could have it scroll once and stop? I've seen that on other radios and haven't hated it.

Thanks!
-Ted

alexsosnovsky, 2022-10-27:
To provide some support to the idea, my Ford Sync screen does basically this for Bluetooth music, where it shows the first x characters that fit on screen, and doesn't scroll further. I don't think there's a scroll option. I honestly never thought much of it until seeing this issue.