tedsalmon/BlueBus #126: Question: Support for the Dialing UI on Navigation Screens

multilan-tarek, 2022-10-17:
Hello!

Quick question: Is support for the dialing ui on navigation screens planned?

Greeting from Germany!

tedsalmon, 2022-10-22:
Tarek,

@nabucho has been working on this feverishly in branch `feature/phone`

multilan-tarek, 2022-10-22:
Great work, that's really cool!

Would it be possible to display messages in the phone menu using the Bluetooth MAP profile? Not that it's really useful, but it would be a cool gimmick.

tedsalmon, 2022-10-30:
> Great work, that's really cool!
> 
> Would it be possible to display messages in the phone menu using the Bluetooth MAP profile? Not that it's really useful, but it would be a cool gimmick.

The BM83 doesn't support MAP, but the BC127 does. Because of this disparity, we will not be supporting SMS readouts on the Nav UI. :(

Keff789, 2022-10-31:
SMS is anyways not useful, if something push notifications like Whatsapp would be cool :p but that seems way out of scope

nabucho, 2022-10-31:
Yes, general notifications (ANP profile) are unfortunately not at all supported by the Bluetooth chips used in BlueBus.