tedsalmon/BlueBus #133: Question: Volume button function

sasalaa, 2022-10-30:
Could the volume buttons on the steering wheel and radio be assigned to control the output level of the BlueBus output not radio output?

Car: E46 2001

tedsalmon, 2022-10-30:
@sasalaa,

No, that's not really possible. Why would you want this anyways? I always recommend keeping the phone volume at maximum. :slightly_smiling_face: 

-Ted

sasalaa, 2022-10-30:
My idea is to keep the phone volume at maximum, send audio from Bluebus straight to my aftermarket amplifier but still retain audio level control on steering wheel and radio trough the bus cable.

tedsalmon, 2022-10-30:
> My idea is to keep the phone volume at maximum, send audio from Bluebus straight to my aftermarket amplifier but still retain audio level control on steering wheel and radio trough the bus cable.

Are you planning on keeping the factory radio? The BlueBus won't work without it (at least not easily, though anything is possible...). How would you switch the audio input source on the amplifier if you ran the line level outputs directly to it?

sasalaa, 2022-10-30:
The factory radio is staying. Amplifier will receive two separate signals. I suspect that when I switch to FM on my radio, BlueBus will pause the music on my phone so it won't be a problem to have 2 different signals running into one amplifier. 
Why I'm even thinking about doing it is to lessen the amount of links in my signal chain and getting the audio straight from BlueBus to the amp without any long wire runs and the radio could possibly also introduce some unwanted noise.