tedsalmon/BlueBus #140: Microphone not picking up any sound

nmckeown51, 2022-11-11:
2001 540i sedan w/ nav, I have the SES jumper installed, I installed a microphone in the headliner and plugged it into the black connector that was tucked above the headliner, I can hear receiving ends of calls perfectly but the microphone does not pick anything up, mic gain is set to 23 decibels. Everything else functions as it should, any help would be appreciated.

tedsalmon, 2022-11-16:
@nmckeown51 

what microphone part number are you using? All I need are the last three digits :)

tedsalmon, 2022-11-17:
@nmckeown51 

ah, you were unlucky enough to get the one microphone that flat out doesn’t work. Please source either of the following from eBay: 84318380338 or 84316938762.

nmckeown51, 2022-11-17:
OK, that you so much!

nmckeown51, 2022-11-29:
So I got a microphone with pn:84316938762 and it's still not picking up anything.

tedsalmon, 2022-11-29:
@nmckeown51 Can you shoot me an email with images of how the BlueBus is connected to your car, please?

closed 2022-11-30 by nmckeown51