tedsalmon/BlueBus #144: Question: Navigation messages on screen

Seilafeh, 2022-11-21:
Hi, is it planed to display navigation prompts e.g. Google Maps on the OEM screen via Bluetooth? Including a little Google maps feature like an arrow or "turn left in 200m" would be a huge improvement in my opinion. I'm not a programmer but there seem to be protocols for this since you can get this kind of information on your SmartWatch or Xiaomi MiBand.

Greetings :-)

tedsalmon, 2022-11-25:
> Hi, is it planed to display navigation prompts e.g. Google Maps on the OEM screen via Bluetooth? Including a little Google maps feature like an arrow or "turn left in 200m" would be a huge [...]

I would love to do this, but I am struggling to find the APIs to achieve it. In fact, this is the main reason I want to create an app and make use of BLE. Alas, I must dig deeper.

Seilafeh, 2022-12-01:
I did some research. I am no programmer but tried my best to find a solution. Here are my findings:

An app will be required

The app needs notification access

The Google maps notification needs to be fetched by the app (could also be done with any other notification)

The app needs to send the maps data to the bluebus

Bluebus needs to display the last information and update it whenever there is an update, but keep displaying it when not.

The Google maps push notification includes ETA, distance left, distance until next turn and direction of turn.

 I hope it helps :-)

Keff789, 2022-12-06:
Yea that is maybe a great idee. To have a BlueBus Android App that fetches Notifications, then we also could show incoming Messenger data (which was a request at one point i think).

There is the NotificationListenerService (https://is.gd/5yKS7p) that maybe should do the trick :)

nabucho, 2022-12-06:
Hi, just curious here - if you have a phone holder on a vent - the screen of the phone is way bigger than headunit, and you can have all the UI for maps and Android Car Mode with big buttons just there on the dash.
Does this not work well enough? (It does for me, but I am trying to understand value, apart from having _comprehensive support of whatever can be supported_). Also, for navigation - you cannot display arrows, just some text information - cannot imagine it would be safe to try to read the text navigation info on that small screen.
I am trying to understand.

Keff789, 2022-12-06:
I dont always have my Phone on the Vent so in that case it would be nice to have the information. Also as i am aware since november 2022 andoird car on the phone is almost dead, you can just use google maps with the drive assistant bull. 

Also the 16:9 navi unit is able to indicate that a turn is approching by showing the meteres to the turn in the instrument cluster, but that would be impossible to emulate correct?

nabucho, 2022-12-06:
To me the biggest question is, what data can you capture on the phone, if you could get this info from the "notifications" the apps send to notification system, if they send them often enough etc. But that is a task for an Android developer to bite into.

tedsalmon, 2026-02-24:
Unfortunately, this is not possible to implement on iOS, even with an App. For that reason, I will have to close this ticket :(

closed 2026-02-24 by tedsalmon