tedsalmon/BlueBus #145: Device cannot be found

TechlabNCK, 2022-11-26:
I've recently begun upgrading an audio system in a 1999 E39 M5. Naturally, BlueBus was the go-to choice for integration. 
Unfortunately, I'm not able to find the BlueBus when I search for new devices. 
I've cleared all the previous devices, then paired my earbuds anew, to make sure the phone is not at fault. 
Tried the same thing with 3 other Android phones.  Still no success. 
It's a brand-new BB, just received Nov 5/22.  First installation for this car, but not the first time I've used a BB. 
The BB powers up just fine, and absent the BB, the car's audio system works fine with the original CD changer plugged in. 
Head unit display still shows 6 CD index.  My phone has been sitting on top of the BB for 20 minutes at a time, patiently waiting for the BB to show up.  It has not yet arrived.............
What am I missing?

tedsalmon, 2022-11-26:
@TechlabNCK 

You have to manually enable pairing through the UI. Do you know what generation Navi you have? You should have a menu by now…

Thanks!

TechlabNCK, 2022-11-28:
Hi Ted, 
The car I'm working on does not have factory Navi. No DVD unit for the map disc. It has a 6 disc CD changer, TV tuner, DSP amp and the radio/preamp box. 
I've connected the BB to the factory CD changer leads, and alternatively connected only the 3 pin power plug. In none of these configurations does the BB show up in my 
phone's "Available Devices" list.  The screenshot shown was taken after the phone was sitting on top of the BB for 20 minutes,  I had one of my iPhone using colleagues attempt to 
connect, with the same result. No BB found. 
Please advise on what I need to check next. 
Hope you had a good Thanksgiving weekend.
Kevin.

[image]
[image]
![IMG_20221
[image]
128_121845](https://user-images.githubusercontent.com/119090430/204214168-51381495-6322-4b24-b15e-4d1c929b2862.jpg)

TechlabNCK, 2022-11-30:
I was able to initiate pairing through terminal. 
There may be an issue with the head unit.  Tried to set the DSP input (analog/digital) and it doesn't show up on terminal. 
I'll look further. 
Any insights would be greatly appreciated!