tedsalmon/BlueBus #146: Setting number of "blinks" does not stick

8one6, 2022-11-27:
2004 E46 Coupe, non-nav

I am able to set the number of "comfort blinks" in the settings menu (confirm: I *do* hit save after making the change).  However, the blinker doesn't actually blink that many times, at least when the engine is running.

If I set the setting while the engine is running, it does nothing at all and I just get one blink no matter what.  If I set the setting when the ignition is in position 1, I do get the correct number of blinks while the ignition remains in position 1, however, as soon as I start the engine, the blinker only blinks once.

Any ideas?

tedsalmon, 2022-12-02:
That makes little sense :(.

Did you email me about this as well? I need to know if you're the same user or if I have two users with the same issue :)

piersholt, 2022-12-02:
That's super weird!

What firmware version are you on?

arrdem, 2023-05-19:
Same issue on an e85m, version 1.3.0, business CD. Will try to figure out grabbing an ibus log file.

piersholt, 2023-05-26:
A log would be great @arrdem. Is the issue occur intermittently, or the blinkers behave the same way all the time?

tedsalmon, 2023-05-26:
I think 1.3.0 had this bug, actually. Try: https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_3_3_rc11.hex

arrdem, 2023-05-26:
Will flash this on tonight. Are there notes on dumping the ibus debug output to a file or otherwise teeing it so as to keep the console usable somewhere?

arrdem, 2023-05-27:
> A log would be great @arrdem. Is the issue occur intermittently, or the blinkers behave the same way all the time?

Blinkers behave the same all the time

arrdem, 2023-06-09:
With the linked 1.3.3 release candidate flashed to the bluebus and checked I still see the same behavior. Sending `SET COMFORT BLINKERS 5` and then testing the comfort blinkers produces the factory default three blinks with and without turning off the ignition.