tedsalmon/BlueBus #149: Question: Can I set the default source/mode to BlueBus/CD ?

mdesilva, 2022-12-21:
I've been using the BlueBus for 3 months now and its been working flawlessly. The only issue I have is, not stemming from the BlueBus itself, is that every time I start my car it defaults to the FM radio instead of the BlueBus/CD source that was selected from my previous drive. Since the volume is usually up, that means that I get blasted with a few seconds of unwanted loud radio/static from whatever radio station is selected until I quickly hit the mode button to switch the source to BlueBus.

My question, how do I set the default mode to BlueBus so that I don't have to select it every time I start my car ? 

This is for a 2005 BMW E46 with Business CD Radio.

tedsalmon, 2022-12-21:
The short answer is that this should not be occurring. Would you be open to providing a log? If you turn the car off, then start it back up within ten seconds, does it stay in Bluetooth mode or does it fall back to FM consistently?

mdesilva, 2022-12-26:
Sure,  I'll grab some logs and also verify whether if its stays in Bluetooth mode after 10 seconds of restarting.

ZHPguy, 2022-12-26:
The Bluetooth will auto play when turned off and on for 10 and even 20 seconds

tedsalmon, 2023-01-28:
@mdesilva

Can you try this? https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_3_2_rc2.hex

-Ted

mdesilva, 2023-01-29:
Hi @tedsalmon, I tried firmware version 1.3.2 RC2 that you linked above. At first all of the functionality of the BlueBus was not working as expected. After disconnecting and connecting the device 4 times everything seems to work, including having the default source as Bluetooth. I will turn on the car again 24 hours from now to ensure that the default source is still Bluetooth (as in the past the default source sometimes persisted throughout the day but not the next day), and then we can close out this issue. Thanks again ! 

BTW, the music from the BlueBus sounds amazing on my factory Harman Kardon sound system !

tedsalmon, 2023-01-29:
@mdesilva 

Glad to hear it's working! You probably didn't disconnect the unit from USB for the update, right? That would cause it to lose sync with the car if the ignition was off until after you finished updating it, and is likely the issue you encountered post-updated.

Thanks!
-Ted

mdesilva, 2023-01-31:
@tedsalmon 

Unfortunately after switching the car on today my radio defaulted to the FM tuner again. So it seems like the mode setting does not persist after some time. Let me know if there is a specific sequence that you want me to follow when collecting logs.

nabucho, 2023-01-31:
From my observations this can happen when radio thinks the CD Changer ( BB ) is not connected / responding soon enough after unlocking the car. Do you have original CD Changer? Could you connect it and test, if the CD Changer mode stays on overnight? If not - it can signal ibus cable connection issues as well. Would explain also the issue with having to disconnect 4 times last time you upgraded. 

However - you also mentioned that your radio does not even keep on CD mode? Can you confirm that when you leave the car while on CD mode, and start next morning, it goes to FM? If so, than there is bigger issue with your radio.

mdesilva, 2023-03-15:
@nabucho Yes it does not keep CD mode overnight when the BB is acting as the CD changer. I will plug the CD changer back in and see if that persists overnight.