tedsalmon/BlueBus #151: Question: Best way to sniff the canbus? And contribute?

Quattro20v, 2022-12-22:
Hello, apologies if posting in the wrong spot. What is the best/easiest way to sniff the canbus? I found this article https://www.instructables.com/CAN-Bus-Sniffing-and-Broadcasting-with-Arduino/ is this the best way, or is there a better way? Would INPA be of any help?

Essentially I am new to programming with cars and would like to start off with implementing some features. For example the fog lights to turn on when I come back home at night and open the door. I've tried digging online for information on what binary value would trigger fog lights when given to LCM but did not find any documentation. So my only guess is to extract it by sniffing.

Is there anything I should be concerned in breaking on the car?

I assume the best way to develop is to just write code, compile and plug into the car to test, rinse and repeat?

Also any useful resources would be helpful, if available.

piersholt, 2022-12-23:
BlueBus does not connect to, or interface with CAN bus. BMWs of this era have multiple buses, all of which serve different purposes. BlueBus deals exclusively with the K-Bus and I-Bus (which compared to CAN, are significantly simpler.)

**Body [Karosserie] Bus (K-Bus)**
> K-Bus was added to the E38 along with the I-Bus. Models without Navigation or IKE will use the K-Bus only. Both of these bus systems are technically identical, the only difference is their use by model.

**Instrument Bus (I-Bus)**
> I-Bus was introduced on the E31 as the information bus. The E31 version of the I-Bus was used for body electronics and driver information systems. With the introduction of the E38, the I-Bus is now referred to as the instrument bus.

Based on your question, it sounds like this is a burgeoning curiosity? If that's the case, I think this project is likely to overwhelm you, and as such, probably not the best place to start.

I'd recommend starting with getting diagnostics up and running. INPA is a good place to start. I'd recommend the BMW forums to learn more about this, as it'd been covered at length.

Following that, I'd then look at [NavCoder](http://www.navcoder.com/), which will allow you to monitor bus traffic, and communicate with the modules via a UI.

Quattro20v, 2022-12-23:
Thank you for explaining this more. I think I was getting lost in all of the systems, from trying to learn online. It's a bit overwhelming learning a new subject from scratch.

I do have INPA set up, so looks like the NavCoder is the next step. I really appreciate you pointing me in the right direction, from reading about it, it looks like exactly what I was missing to be able to learn more I-bus and how all the systems work.

I do have a few years of software development experience, just nothing involving cars. I am eager to learn though, just need to put all the right pieces together to get a base (Which I think NavCoder will help immensely with!). Thanks again!

sleuth255, 2022-12-23:
You'll also need an I-Bus interface to go with NavCoder.  The software includes an electronic schematic for one that you could construct if you're handy with a soldering iron and have components laying around,  or you can purchase a usb adaptor from the author's company in Germany for ~$60.  I actually went this route myself and now have it permanently wired into my trunk.  It's 25 euros to fully unlock NavCoder and it can take a while to get your activation code back since the author doesn't check his requests very often.  The listed email addresses are also out of date.  HIs current email is jbsiegenthaler@gmail.com so you could ping him there if he doesn't see your activation request.

piersholt, 2022-12-24:
@Quattro20v Happy to help 😄 

I'd also recommend searching for the following BMW documents. A few exact keyword searches should yield results on Google 😉 

They'll give you an overview of the various bus systems- their purpose, how they function, how they interconnect etc. The first is more pertinent to the generation of vehicles that BlueBus applies to, but you may still like to read the second.

**1. Introduction to Bus Systems**
> [image]

**2. Bus systems**
> [image]

piersholt, 2022-12-24:
@Quattro20v ah! Yes @sleuth255 is quite right. A plug and play interface for I/K-Bus will be useful. However, you can still try NavCoder with your diagnostic interface. Also, unless it's since changed, you should be able to try NavCoder without a licence, it will just lack some features.

Quattro20v, 2022-12-24:
@sleuth255 Thanks, I'll order the interface so I can play around with it! From looking at it, it looks like it comes with a connector that then has ~4 wires. Is there a plug and play wire/connector option to purchase on top of that, to hook in instead of cd changer? I'd rather not do any splicing if I don't have to. Or what is the best route that you would recommend? I checked the installation documentation for the interface, but couldn't decide what the best option is for the purposes of connecting the car to a laptop

@piersholt Thanks I appreciate the resource drop, can't wait to absorb all this knowledge! 😄 It's pretty fascinating how all this works, I wish I started looking into this earlier!

sleuth255, 2022-12-24:
Yup, they also sell a harness that plugs into the CD changer if you don't want to splice that USB connector into the CD changer cable itself.  Its a bit tricky to locate so here's a link:
https://shop.ibus-app.de/i-bus-app/zubehoer/adapterkabel/196/adapter-fuer-resler-modul-zum-zwischenstecken-am-cd-wechsler/wechsler-vorbereitung

As @piersholt indicates, NavCoder will work without a license but stuff like LCM coding won't be active.

tedsalmon, 2026-02-24:
Closing stale issue.