tedsalmon/BlueBus #152: Will Not autoplay

ZHPguy, 2022-12-25:
Hello, I have a 2004 E46 convertible and love the bluebus.  However, it will not automatically connect when I start the car.  I have to press the “mode” button to switch it to bluebus.  Then it works perfectly.
Also, I thought my car came with a mic.  It obviously does not as no one can hear me when on a call.

tedsalmon, 2022-12-26:
This sounds related to #149

ZHPguy, 2022-12-30:
It does auto play after turn car off and back on in 10 or even 20 sec

tedsalmon, 2023-01-05:
@ZHPguy

@nabucho recently found this typo: https://github.com/tedsalmon/BlueBus/blob/master/firmware/application/handler/handler_ibus.c#L154

I think this might be related to what you're experiencing.

-Ted

ZHPguy, 2023-01-05:
Thanbk you.  How do I update my firmware

tedsalmon, 2023-01-28:
@ZHPguy 

https://www.bluebus.dev/software

But use this version instead: https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_3_2_rc2.hex

Thanks!
-Ted

closed 2026-02-22 by tedsalmon