tedsalmon/BlueBus #154: E39 MID song metadata not updating

ExCurryBur, 2023-01-13:
Hi, 

My E39 has a MID with DSP.
After updating to SW 1.3.0, the song name is not updating on the MID.
It will get stuck and display as "Paused" on the screen after going to next song.
When pressing TEL again it will update the screen with corect metadata. 

Thanks,
Wayne

tedsalmon, 2023-01-20:
@ExCurryBur

Do you have a V1 or V2 unit?

-Ted

ExCurryBur, 2023-01-20:
A v1 unit. 
just to clarify I have to press the phone icon next to the tel button to refresh, not the Tel button. 

Thanks

tedsalmon, 2023-01-28:
@ExCurryBur

Can you test this firmware out? https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_3_2_rc2.hex

Can you also share what media player you're using?

Thanks!
-Ted

ExCurryBur, 2023-01-29:
Thanks!
I tried this firmware and it fixed the problem. 
I am using YouTube music app on iphone.

tedsalmon, 2023-01-29:
Thanks for verifying @ExCurryBur !

closed 2023-01-29 by tedsalmon