tedsalmon/BlueBus #156: Unable to upgrade firmware

Dashdot66, 2023-01-28:
As a fan of updates I tried to upgrade to the latest version 1.3.0. The firmware tool however is unable to connect within 10 seconds.
[image]

The cable is the same one I used for previous versions, I left it in the car because the bluebus is not easily accessible.
I am able to open a terminal on the bluebus:
[image]

The green light is on and no Philips screws to loosen:

[image]

Any thoughts?

tedsalmon, 2023-01-28:
@Dashdot66 

In the terminal, run "bootloader" and then try the tool again. Some change I made at some point impacted the ability of the tool to place older software versions into bootloader mode on macOS machines.

Dashdot66, 2023-02-20:
That did work, I was able to update the version and the loader confirmed version 1.3.0 however the bluebus was lost after this point, no selection in the screen no Bluetooth and no terminal only the load tool is able to connect. What is smart to do? Try again or load version 1.1.18? I really don't want to brick it.