tedsalmon/BlueBus #159: Distortion at high volumes

nmckeown51, 2023-03-23:
I recently did a Bavsound upgrade on my 01 540i. Now that I can play music at high volumes without the speakers rattling away, I noticed apparent distortion over the Bluetooth connection using the BlueBus. Tapes sound great at high volumes so I know the speakers and amp is okay. I have the dB on the audio tab set to -10 and I can lower it more but it doesn't seem to make much of a difference and the feedback from the alternator becomes way more noticeable. Is there anything that can be done to improve the sound quality? or is it just the limitation of the hardware? Just looking for some suggestions before I go back to burning CDs XD. I've tried re-pairing the device in case it was just a bad pairing profile but it made no difference.  I've also tried other eq settings but none of them make that much of an improvement.

tedsalmon, 2023-03-23:
@nmckeown51 - Could you please email support@bluebus.dev?

tedsalmon, 2023-04-07:
Closing -- handled over email.

closed 2023-04-07 by tedsalmon