tedsalmon/BlueBus #160: E39 - Comfort Functions Not Working

multilan-tarek, 2023-03-31:
Hi!

I have basically the same issue as #158 with my 2001 E39 525i. Turn signals work, but auto-unlock does not.

Also I compiled the latest version of the feature/autotime branch and tried it, but no change (PDC isn't working either, but that's probably because it's not finished yet?)

How can we debug this?

The car has a GM3 High Module

tedsalmon, 2023-04-01:
@multilan-tarek,

based on your name, your BlueBus was delivered late last year, right? The auto time feature will never work for you. It’s only for HW 1.x BlueBus units and for iPhone users. This is because the BM83 doesn’t support AT commands, and Android doesn’t respond to AT queries.

as for your unlock issue, can you please send me a log of you going 10 km/h?  I want to see if it’s even trying to lock your doors.

-Ted

multilan-tarek, 2023-04-01:
I sent you my file via Mail.

Autolocking is coded in the GM3, but the ibus locking will also notwork either. (Even with coding removed)

tedsalmon, 2023-04-07:
After reviewing the log file, I can see the correct messages are being sent by the BlueBus, so I suspect it's an issue of the GM3 variant not being supported.

multilan-tarek, 2023-04-08:
Hi Ted, I took a picture of my GM3 and sent it to you via Mail.

multilan-tarek, 2023-04-08:
Could it be that US models didn't have Servotronic? Because GM3 High was installed only in cars that were equipped with it.