tedsalmon/BlueBus #164: Bluebus not switching off

dertimaushh, 2023-04-17:
[image]

Hi Ted,

I think I've found the root cause for my battery drainage. Bluebus isn't turning off. 

Everything works fine expect the fact it will not switch off.

Tried FW 1.30 and 1.32 now. 

Any thoughts? 

Best regards,
Tim

tedsalmon, 2023-04-17:
@dertimaushh ,

Are you sure? With the keys out of the ignition and the doors closed, the LED on the BlueBus should go out within three to five minutes.

-Ted

dertimaushh, 2023-04-17:
Yes, after some days of quiescent current measurement - I can confirm it's never turning off.

tedsalmon, 2023-04-17:
@dertimaushh 

But is the LED going out? How much current is it pulling?

have you tried running the restore command? Auto off is controlled by a configuration byte.

-Ted

dertimaushh, 2023-04-17:
~ 0,02-0,03A

Havn't tried any commands yet.

dertimaushh, 2023-04-17:
[image]

Now let's wait.

dertimaushh, 2023-04-17:
Success - restore solved the issue :-) 

[image]

closed 2023-04-17 by dertimaushh

ShonkyCH, 2023-04-17:
> The Led stays on all the time.
> 
> ~ 0,02-0,03A
> 
If that's the total current measured (which it looks to be with the negative disconnected ie not the BlueBus alone) with the car asleep 20mA-30mA is well within spec (<40mA). For reference mine sits at about 19-20mA (with a sleeping BlueBus)

Seems like you've fixed a problem still.