tedsalmon/BlueBus #166: LCM not working correctly

kaikoehler99, 2023-04-19:
When the blusbus is connected the LCM/gauge cluster dimming of the display/indicators isn't working.
Also I am missing the option to swap settings in which model the bluebus is connected: I have an E39 and an E46. Or was this removed in a software update?

tedsalmon, 2023-04-20:
@kaikoehler99

Are you making use of Parking Lamps? Turning those off will fix your dimmer issue.

The vehicle type option was removed. Getting this information from the cluster is fairly easy, so there's no need for user input.

-Ted

kaikoehler99, 2023-04-20:
Thank you for the quick answer! This indeed did help fix my dimmer issue :)
Is there a way to get this work correctly or is the limitation the LCM in the E46?

tedsalmon, 2023-05-08:
> Thank you for the quick answer! This indeed did help fix my dimmer issue :) Is there a way to get this work correctly or is the limitation the LCM in the E46?

Not entirely! The LSZ & LCM do not process dimmer pot adjustments while you have diagnostics running. I have some ideas, but certainly none would work for the LSZ found in the E46, E83, E85 and E86 as sending values for the backlighting does trigger a change in the cabin lighting (therefore I cannot send dummy values, like on the E38, E39 and E53).

Thanks!
-Ted

pyrophan, 2023-11-17:
V2 with 1.3.3 here to confirm. Adjusting the dash light dimming can cause erratic HVAC fan speed and may result in the cut-out of A/C.

No use of parking lamps.
Only use of one touch turn signals.

tedsalmon, 2023-11-17:
@pyrophan 

what year and chassis vehicle?

pyrophan, 2023-11-17:
1999 E39, non-dsp

pyrophan, 2023-11-29:
I have heard of it, but have never used it. I'm sure it's something I could learn to use with a little guidance. I already have a USB to K+OBDII cable and the OBDII to underhood round pin adapter. 

Where can I find INPA?

tedsalmon, 2026-02-25:
The issue of the IHKA being impacted by lighting features can be mitigated with this command as of 1.3.5:

```
SET LMPOLL OFF
```

closed 2026-02-25 by tedsalmon