tedsalmon/BlueBus #168: can't find device

avusm3, 2023-05-22:
Hi all, 

I have had bluebus for awhile.  2004 M3 with NAV.  I cleared my phone at the pairing menu, and now, bluebus cannot pair with my phone.  Had worked fine before.  Updated the firmware and that didn't do anything. Any ideas?  Any other way to pair than from the NAV?

tedsalmon, 2023-05-22:
@avusm3 

Did you try unpairing it on your phone side as well and then rebooting the phone?

iOS is notorious for not "seeing" a device that you recently unpaired.

Also, did you enable pairing on the BlueBus end?

-Ted

avusm3, 2023-05-22:
Yes, it does not appear on my phone side list at all, and I tried rebooting the phone too.  It's been awhile since I unpaired it.  Pairing was enabled on the Bluebus end. Any way to pair it from a computer or something like that to get it going again?

ShonkyCH, 2023-05-22:
Getting the BB into pairing mode by any other method won't make it operate any differently.

Do you have, or can you borrow another phone or tablet to see if the BlueBus is transmitting? Maybe trying pairing that device temporarily. That would essentially rule out the BB as a problem.

avusm3, 2023-05-22:
I will give that a shot to confirm....but given that the phone I'm using is pairing with other cars and devices without issue, I don't think it's the phone....

avusm3, 2023-05-26:
I tried another phone, didn't work. Anyone have any other ideas?

tedsalmon, 2023-05-26:
> I tried another phone, didn't work. Anyone have any other ideas?
> […](#)

Send me an email at support@bluebus.dev -- you'll need to send it to me for repair. :(