tedsalmon/BlueBus #169: Blueblus aktiv -> GPS Signal weg

timo330d, 2023-05-22:
Hello all,

my Bluebus V2 does not like the way I want. :)
In Bluetooth streaming, the car keeps losing the GPS signal. As soon as I switch to radio AUX or CD (E46 Bordmonitor with CD drive) the signal is back and remains constant. As soon as I switch back to the Bluebus Wechsle the GPS signal disappears within a few seconds.

I tested it with another MK4 and another GPS antenna. As well as another Bluebus version 1.3.2 My cell phone is a Samsung S21.

Bluebus V2 40/22
S/N 1003

Would be glad about a help.
Many greetings from Germany

tedsalmon, 2023-05-26:
Can you possibly show me a video of what you mean by "losing GPS signal"? Typically there's a number of pixels in the satellite box to indicate GPS signal, but I'm not sure we're talking about that...

timo330d, 2023-05-26:
https://youtu.be/z1z1xm3NWN4

timo330d, 2023-05-29:
Hello Ted,

I have now changed something else. Namely, I have now integrated the Bluebus into the back seat. There, where normally the ski bag is installed. The GPS signal has become much better with this change. But still not quite satisfactory. What I noticed is that when I put the phone next to the Bluebus, it fits perfectly. So the Bluebus must be causing some interference. 

Thanks already

tedsalmon, 2023-05-29:
@timo330d 

That's kind of nuts -- no one else has ever reported an issue with the GPS signal being interfered with by the BlueBus. Can you post some pictures how the BlueBus is connected? Have you ensured the GPS input into the back of the Navi is solid?

-Ted

timo330d, 2023-05-29:
@tedsalmon 

Hello Ted,

yes, that's crazy too. Have a few pictures here. As you can see the MK4 is relatively new, also already have a new GPS module installed, as well as a new antenna. That one can rule out the error here already times. Furthermore, I have yes the Bluebus as far as the changer cables allow, laid to the front. Of course, I then had to extend the cable for the micro.

If I now put the phone on the middle arm of the back seat, which is only a few cm between Bluebus and phone, I have no loss of GPS satellites.  If I have the phone in front next to the screen, I'm already missing a few satellites. In radio or AUX mode 6 pieces in Bluebus streaming 3 to 4 pieces.
[image]
[image]
[image]
[image]

tedsalmon, 2023-06-06:
@timo330d 

To be honest with you, I don't know what to suggest other than trying a different unit. I've never had anyone else experience this, though I assume only a small subset of customers even use the Navi.

Can you email me at support@bluebus.dev?

Shanee46m3, 2023-06-23:
Not a big issues, but I validated. When in radio mode, position known.  When in bluetooth, it stays in satellite search. 

[image]
[image]

Shanee46m3, 2023-06-23:
Makes no sense though. Bluetooth and GPS are on different frequencies.   Bluetooth is 2400 MHz to 2483 MHz while GPS is between 1100-1600 (can’t remember exactly. If Bluetooth were the issue, the oem TCU would cause issues.   There are frequencies that can interrupt gps though.  781-794 MHz and 521-530 MHz.  It seems by doing an AM radio scan, something is broadcasting on around 790

https://github.com/tedsalmon/BlueBus/assets/137462883/b18b9e8c-2a5e-47cb-aadb-43c42899b00b

ShonkyCH, 2023-06-23:
AM is kHz though. A long way away from 1.57GHz (L1 GPS)

tedsalmon, 2023-06-23:
I have to try this tomorrow. My M3 can’t roll outside right now, and my Touring doesn’t have a GPS antenna because… well, Touring trim is expensive and I didn’t feel like breaking a bunch of it to install an antenna for a system I’ll never use 🤪.

I’ll slap an antenna and test both the v1 and v2 hardware.

My assumption is the BC127 isn’t well shielded or is otherwise so noisy that it’s bleeding over (though over a GHz is a stretch).

In my late teens I worked at a WISP that was licensed on the 2.5GHz spectrum and I remember a case where a certain Air Force base (Wichita Falls, TX) forced us to turn down some of our cells because they were somehow jamming GPS operations at the base. RF is dark magic 🤷🏻‍♂️

Shanee46m3, 2023-06-23:
Yes.  They are noted as frequencies that can potentially interrupt gps.

Shanee46m3, 2023-06-23:
@tedsalmon 

Not a big issue for me, i am just geeking out.  My GPS disk is 2007 😂.    I am just freaking thrilled I am no longer stuck with the same 12 song cd I have been listening to for 6 months.  

Actually thinking out loud.  The oem antenna if equipped is under the rear ashtray on the center console.   Wonder if BMW ran into something similar.

tedsalmon, 2023-06-23:
@Shanee46m3 

I’m thankfully my cars always at least had AUX, lol. 

Out of curiosity, are you a new user? Does your unit have a green or blue LED?

With regards to placement of the factory BT antenna, this is a possibility. I always assumed it was placed there so as to provide the best quality to the user (since it was literally smack dab in the middle of the passenger compartment).

-Ted

ShonkyCH, 2023-06-23:
Maybe but I doubt it. My BC127 based unit sits directly on top of the MKIV (M3 coupe) and GPS is fine. I assumed BMW's choice of location is simply to be close to the driver.

Shanee46m3, 2023-06-23:
@tedsalmon 

yeah, but its a pain to charge my phone and aux at the same time.   Too many wires.  

New user blue light.

tedsalmon, 2026-02-25:
Closing this one -- As much as I would like to get to the bottom of it, that's not really feasible as I can't alter the antenna design of the unit (since I use a modular BTM). 

I have noted that GPS begins to work about a minute after setting off and then stays working, so hopefully that's the same for everyone.