tedsalmon/BlueBus #171: E39 - Can't Switch Audio Mode to Bluetooth

SmoglessPrune, 2023-06-26:
Hello,

I received my BlueBus a few weeks ago and I'm having issues getting it working with my '02 E39 M5. The car is wired for a CD changer but does not have one installed; it looks like the CD changer was forcefully removed by a previous owner because the connector on the 3 pin power was ripped off and I had to put a new end on it. My car does not have a phone so I only have the 3 pin power and 6 pin analog DSP cables connected to the BlueBus. The blue LED lights up and the radio works as normal, but when I hit the mode button I only get radio or tape, it wont switch the input to bluetooth. I tried updating firmware on the BlueBus and that was successful but the issue didn't change. Something seems to be stopping the BlueBus from communicating with the car, but I don't know if the issue is in the BlueBus, my radio, or the wiring from whatever happened to the original CD changer. Any troubleshooting advice would be much appreciated.

tedsalmon, 2023-06-26:
@SmoglessPrune,

Is the light on the unit on constantly when the ignition is on? A good test is to leave the ignition on and then check if the light is still on after 120 seconds.

P.S. Email is better for these types of support requests :smile: 

-Ted

SmoglessPrune, 2023-06-26:
Thanks for the tip, I will send an email next time.

Yes, the light is on constantly with the ignition on. I left the ignition
on for 120 seconds and the light was still on solid