tedsalmon/BlueBus #177: Bluebus not visible on head unit.

abrahammathai, 2023-09-13:
I was using bluebus for 3 months now. All of a sudden it went off. Phone says its paird to bluebus. But nothing of bluebus is visible in the mid unit of bmw e38 raio

tedsalmon, 2023-09-13:
@abrahammathai 

can you unplug it and plug it back in? Shoot me an email so I can send you the latest software build

abrahammathai, 2023-09-15:
When disconnected the battery and reconnected  its ok now. Kindly send me latest software build. I cant hear the phone well. Does changing   mic make any change?

Thanks

abrahammathai, 2023-09-15:
My e mail is     pulinattu@hotmail.com

abrahammathai, 2023-09-19:
Good morning, Ted:

I have a 1999 BMW 740IL. I installed Blue bus. Its working good. Occasionally it doesn't work. When that happens if I disconnect battery and connect again it's ok.  My biggest challenge is the audio during telephone call is really bad.  From the dealer I found out my microphone in my car is part number, 84318380338.  If i  change my microphone, will it make a difference. If yes what part number, I should use.  I have one available, that is used and its part number is 84316938762.  Could you also send me the link with instructions to update the software. Is there a aftermarket microphone I can use?
Thanks in advance

Abraham Mathai

tedsalmon, 2023-10-13:
Email sent :)

closed 2023-10-13 by tedsalmon