tedsalmon/BlueBus #178: BM53? Europe had the BM54

xXUnbekanntXx, 2023-09-30:
https://github.com/tedsalmon/BlueBus/blob/1bc7e2ae937bdfa3140a62cd753126c614bbea01/firmware/application/ui/bmbt.c#L70

I live in Europe does this line of code effect something? I mean I have the BM54 but if the bluebus is searching for a BM53 it will never find it.
Regards

tedsalmon, 2023-09-30:
@xXUnbekanntXx - This isn’t a problem. The BM54 is just a BM53 with TP/TMC features. :)

closed 2023-09-30 by tedsalmon

xXUnbekanntXx, 2023-09-30:
ok thanks. Then I dont know why I get the UART Error when checking "set log sys on". Could you help me with that one?
On my e39 I didnt even get a picture on the nav screen.

tedsalmon, 2023-09-30:
> ok thanks. Then I dont know why I get the UART Error when checking "set log sys on". Could you help me with that one?
> On my e39 I didnt even get a picture on the nav screen.

This is normal when you reboot the unit. Are you seeing ibus traffic? Can you shoot me an email with pictures of the UI your car has?

xXUnbekanntXx, 2023-09-30:
ok, I see the ibus traffic from the car side. The "get vin" com also works. I dont have a pic from my own nav screen right now, but basically it is this one. Split screen is active.
[image]

tedsalmon, 2023-09-30:
> ok, I see the ibus traffic from the car side. The "get vin" com also works. I dont have a pic from my own nav screen right now, but basically it is this one. Split screen is active. [image]

That doesn't help, unfortunately. Is there not a "Bluetooth" or "CD 7-01" mode?

xXUnbekanntXx, 2023-09-30:
No it’s not. Only when I connect the bmw cd changer.

tedsalmon, 2023-09-30:
Can you shoot me over a log of the I-Bus traffic? https://github.com/tedsalmon/BlueBus/wiki/Gathering-a-Log

xXUnbekanntXx, 2023-09-30:
[bluebus log.txt](https://github.com/tedsalmon/BlueBus/files/12775330/bluebus.log.txt)
thats it.

tedsalmon, 2023-10-01:
> [bluebus log.txt](https://github.com/tedsalmon/BlueBus/files/12775330/bluebus.log.txt) thats it.

Did you build this unit yourself or get it from me? I would guess that there is a loose pin on the I-Bus transciever based on the amount of General and Frame errors in the logs.

xXUnbekanntXx, 2023-10-01:
I bought a e39 and it was inside. The owner told me that it worked but after a while it stopped. 
Is there a specific pin I could check?

tedsalmon, 2023-10-02:
@xXUnbekanntXx 

Could you please send me pictures of the circuit board? support -at- bluebus -dot- dev

xXUnbekanntXx, 2023-10-02:
[image]
Should I send it to you via Email?

tedsalmon, 2023-10-13:
> [image] Should I send it to you via Email?

Please shoot me an email so we can talk about getting it repaired :)