tedsalmon/BlueBus #180: Voice works in calls but not apps?

DougieMcBunBuns, 2023-10-10:
Hello! I partially bought my Bluebus because chintzy Amazon radio voice options weren't stellar and I tend to listen in on meetings during my commute. However my Teams app doesn't seem to like my Bluebus? It plays the call audio just fine but doesn't pick up my mic audio. Afterwards my wife called, who heard me perfectly loud and clear.

Is there any inherent lack of compatibility with apps like Teams or Zoom with conferencing? Teams just sucks with audio bugs anyway, but I didn't see any issues about Teams specifically so I figured I'd ask for future people searching around about this. I did see some threads about the CVC license, but again, that sounds like it also affects regular phone calls.

Any help is appreciated - thanks all!

nabucho, 2023-10-10:
There should nothing really be linked to BlueBus. I can positivelly say I have spend hours talking on Zoom in my car with Bluebus.

Some apps have options to select which microphone is used, so may be you could try to looking at that, checking the mute in app that could be set by default, etc. 

The fact that phonecalls work well indicates the phone knows there is mic, the hardware is working as well. 

Are you using iOS or Android? Can you screenshot a screen of the conference app with the controls and audio-devices selection when it is not working? 
Can you try WhatsApp, Telegram, Zoom calls? They are are apps and they all work fine with BlueBus, so I would focus on checking the setup of Teams if that is the only app not working.

tedsalmon, 2023-10-13:
@DougieMcBunBuns 

@nabucho is pretty much spot on. I also suspect that Teams is refusing to output via the BlueBus for some reason.

CVC License is only relevant for "Green LED" BlueBus units, and it being missing would result in phone calls not working over BT either.

tedsalmon, 2026-02-22:
Closing stale issue.