tedsalmon/BlueBus #181: Whining or Ground Loop noise on calls and music

mbt28, 2023-11-15:
Hi Ted,

I have the V2 board with latest firmware. I hear the alternator noise from the speakers, I can even listen convertible servo sounds from speakers :) . This problem exists when I choose bluebus as a source, fm and others are fine.

How can I resolve this issue without attaching additional ground loop isolator? I think the board is okay but something wrong with my wiring.

Thanks

pyrophan, 2023-11-17:
Can confirm same issues with V2 on latest firmware in a 1999 e39 without DSP. 

Only happens when the vehicle volume is turned up a lot. Usually turning up the Bluetooth source volume up allows use of a lower vehicle volume and avoids the issue.

tedsalmon, 2023-11-19:
@mbt28

What's the aversion to the ground loop isolator?

Typically, the root cause of this is issues with the rectifier diodes in the voltage regulator for the alternator, whence why the whine follows RPM. PWM from the throttle actuator can also introduce this noise.

In both cases, the GLI is the easiest fix :o

@pyrophan

This is pretty normal -- you're probably hearing I-Bus interference, which will sound like digital clicking noises, rather than a whine that follows RPM, right?

Thanks!
-Ted

mbt28, 2023-11-20:
Hi Ted,

But why I don't have it if I listen FM radio but only in Bluetooth mode? Even if I don't listen something I can hear the hissing noise from the speakers. I doubt I have a problem with my ground wiring. My ground is wired from chassis but there is also telephone audio input ground which is also wired to chassis on head unit stereo. Can the both ground connection from different part of the car cause the issue?

Regards,
Mehmet

StefPedro, 2023-11-24:
Hi, i have the same issue on my 1999 523i. I've had the issue on my Business CD (CD43) made by Blaupunkt, but have retrofitted original navigation with BM54 and HiFi (not DSP) and i hear clicking noises at high volumes, especially in quiet parts of songs.

tedsalmon, 2023-11-24:
> Hi Ted,
> 
> But why I don't have it if I listen FM radio but only in Bluetooth mode? Even if I don't listen something I can hear the hissing noise from the speakers. I doubt I have a problem with my ground wiring. My power ground is wired from chassis but there is also telephone audio input ground which is also wired to chassis on head unit stereo. Can the both pwr and tel ground connection from different part of the car cause the issue?
> 
> Regards, Mehmet

The A/C whine is bleeding into the power input on the BlueBus, then getting piped back out the line level audio output -- this is why it doesn't exist in other audio modes. Different ground levels could cause issues, but the most likely answer is the rectifier diodes in the alternator's vreg. Can you expound on why a ground loop isolator isn't an option? It's the cheapest, easiest fix :)

tedsalmon, 2023-11-24:
> Hi, i have the same issue on my 1999 523i. I've had the issue on my Business CD (CD43) made by Blaupunkt, but have retrofitted original navigation with BM54 and HiFi (not DSP) and i hear clicking noises at high volumes, especially in quiet parts of songs.

Version 1 or Version 2 BlueBus? If you do not know, what color is the LED on the unit? Blue means it's a v2.

If it's a V1, you need to make sure that your phone volume is all the way on max while streaming.

However, what you're experiencing is normal, it just shouldn't be audible over road noise and the music itself while driving.

What you're hearing is impulsive interference from the I/K-Bus that is bleeding into the line level audio input from the CD Changer into the radio.

Thanks!
-Ted

StefPedro, 2023-11-24:
> Version 1 or Version 2 BlueBus? If you do not know, what color is the LED on the unit? Blue means it's a v2.
>
>
>
> If it's a V1, you need to make sure that your phone volume is all the way on max while [...]

It is a V2. I was worried that I had something installed wrong, but if it's normal it's ok. When driving I don't hear it, only when stationary and listening loudly.

redssu, 2024-01-04:
Hey, 

I also have the V2 board and I also hear the alternator whining, how can I connect a ground loop isolator to BlueBus to reduce the whining?

SecT0uch, 2024-01-29:
If anyone is interested here is how I managed to install a ground loop isolator:

I bought this model: [Europe](https://www.amazon.fr/dp/B00BBE59WS?psc=1&ref=ppx_yo2ov_dt_b_product_details)   [US](https://www.amazon.com/PAC-SNI-1-3-5-3-5-mm-Isolator/dp/B001EAQTRI/ref=sr_1_3?crid=5Q1ZZJ2E6BV2&keywords=pac+ground+loop+isolator&qid=1706525537&sprefix=PAC+groun%2Caps%2C269&sr=8-3)

The case is glued together, but I managed to open it.

Do not trust the wiring color code inside, I tested with a mutimeter and here is the correct wiring:
- Yellow: Ground
- White: Right
- Red: Left

Then you can desold the wires, cut the cables of the adaper and sold them back-on the GLI.
Double check the corresponding wires with the help of this image:
[image]

The result should look like:
[image]
(Bluebus connector on the left side)

[image]

Manozco, 2025-08-09:
Hi guys, 
Sorry to revive this (old) thread but I might need help regarding the whining issue. 
I have a bluebus for a 528i E39 (1997 - EU) and it works quite well except that there is this whining noise sometimes. 
I read the thread, and I bought this device on Amazon:

[image]
[image]

(It looks quite similar to the one shown in the previous message, except the soldering part because I have no soldering iron)

However, my issue stands with the connector on the BMW side, it looks like I have 5 cables connected in the Analog Audio part instead of 4 and I'm not sure what to do with the cable plugged on the pin 3:

[image]
[image]

What do you guys think of this ? 

Thanks, 
Manu

tedsalmon, 2025-08-09:
> What do you guys think of this ?
> Thanks, Manu

Pin 3 is a shield — I left it floating on the BlueBus, so you’re free to do the same.

Thanks!
-Ted

Manozco, 2025-08-09:
So I can just plug 1,2,4,5 and put some tape around the 3, right ?