tedsalmon/BlueBus #182: E39 Navigation metadata is not complete and appears like a menu

StefPedro, 2023-11-24:
I have an issue with the dashboard, in the sense that it dosen't show the entire song name, album, artist if their lenght is too big. Also, it dosen't look the same as Ryan's form E39 Source, and others i found online. It looks like a selectable menu. Also, the temp readouts sometimes don't show up.
The car is a European 1999 E39 with retrofitted original navigation. MK3 computer,TV, BM54 and HiFi.
This is my car (sorry for the crappy picture):
[image]
Here's a picture  i found online on how it looks:
[image]

tedsalmon, 2023-11-24:
It looks like your Nav doesn’t support static display mode, which isn’t uncommon as it was only available on the MKIV Nav with late software revision.

Could you supply a picture of your Settings menu? The Navi one, not the BlueBus one.

-Ted

StefPedro, 2023-11-24:
I have the MK3 nav, with its latest software (3-1/63)(V23), that might explain why it's not working that way, and might motivate me to buy a MK4. Thanks for the info!
[image]

tedsalmon, 2023-11-24:
> I have the MK3 nav, with its latest software (3-1/63)(V23), that might explain why it's not working that way, and might motivate me to buy a MK4. Thanks for the info! Settings menu pictures will come tomorrow, since it's night time in Europe. [image]

If you know it's a MKIII unit, then that is the answer I was looking for :). I believe v27 (1/70) is where they added the static display mode, which was implemented to allow the MP3 capable CD Changer to display ID3 tags.

Thanks!
-Ted

StefPedro, 2023-11-24:
> If you know it's a MKIII unit, then that is the answer I was looking for :). I believe v27 (1/70) is where they added the static display mode, which was implemented to allow the MP3 capable CD [...]

Ok, thanks for the info, I am getting a MK4 next week, excited for newer maps, perspective and the metadata showing properly. I love the BlueBus so far!

closed 2024-01-02 by tedsalmon