tedsalmon/BlueBus #184: Install bootloader

leewi, 2024-01-12:
Hello,

Could somebody help me with how to install the bootloader on a just-assembled board?

Long story short, I've built a bluebus for myself :)
I've checked with USBview in windows, the device is connected and FT231X USB UART is recognized.
Tried Microchip's Unified Bootloader host app, but didn't succeed :(

I assume until the bootloader is not flashed, nothing will work, but I've run out of all my ideas on how to get it done.

Thank you in advance,
Levente

tedsalmon, 2024-01-12:
@leewi,

You need to flash it using a PICKIT and the Tag-Connect TC2030 6-pogo-pin header. You will also need to provision the BM83 Bluetooth module. https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/BlueBus_Provisioning_1_3_2.zip

tedsalmon, 2024-01-28:
@leewi,

I believe the PIKit 3 and 4 will work fine. I do not know about the 2.

Thanks!
-Ted

pazi88, 2024-05-06:
I can confirm that PicKit2 doesn't work, even though it should work on PIC24's. PicKit3 works at least.

Chris6626, 2024-09-05:
> @leewi,
> 
> You need to flash it using a PICKIT and the Tag-Connect TC2030 6-pogo-pin header. You will also need to provision the BM83 Bluetooth module. https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/BlueBus_Provisioning_1_3_2.zip

Hi, im currently trying to set up a new BT module. How do I provision the module? Does it have to be removed and connected to the computer via UART? Or is it possible to flash it on the BlueBus while it's installed? Thanks!

Chris6626, 2024-09-06:
@tedsalmon 
> You need to flash it using a PICKIT and the Tag-Connect TC2030 6-pogo-pin header. You will also need to provision the BM83 Bluetooth module. https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/BlueBus_Provisioning_1_3_2.zip

Hi, im trying to provision my BT module, but always end up getting the Error "Failed to execute Script console_firmware_tool".

Heres what i did

-flashed bootloader with pickit3
-flashed Firmware 1.3.0.hex with Firmware Tool V1.4
-unplug USB, Switch to BOOT, Plug USB in, Led blinking
-execute bt_dfu.bat --> Error

Console communication via PuTTY is ok, but BOOTLOADER command does not set mcu to boot when switch is in APP mode.

Am i missing something here?

Thank you in advance!

tedsalmon, 2026-02-18:
Closing stale ticket.

closed 2026-02-18 by tedsalmon