tedsalmon/BlueBus #193: Simulator development?

ticklemonster, 2024-02-21:
Is it possible to run up the bluebus application on the simulator in MPLab X?

I was hoping to capture (with a terminal) and replay messaging (using UART injection), rather than going out to the car to test every time.

With both BlueBusApplication and BlueBusBootloader connected to simulator, the sim halts in the EEPROMSend.
I'm assuming something needs to be set up in MCC?

tedsalmon, 2024-02-21:
I'm not sure the juice is worth the squeeze here -- we would have to emulate the I-Bus traffic for it to be usable, and that would be in the very least a lot of work.

Have you explored the `set ign on` command while not connected to the vehicle?

Thanks!
-Ted

ticklemonster, 2024-02-21:
ibus is also coming in via a UART, right? I assume the same logic would work (capture in the car, replay in the sim).
It would be awesome to connect it with ibusanalyzer for offline development - but that's probably a step too far.

I can debug in the car with the laptop, but I'm lazy.
Collecting physical components and wiring them up on a bench would be the alternative - but that doesn't seem much easier.

Thanks for the cool project, Ted!

ticklemonster, 2024-02-21:
Accepted - there is no plan for this.

closed 2024-02-21 by ticklemonster

tedsalmon, 2024-02-21:
> ibus is also coming in via a UART, right? I assume the same logic would work (capture in the car, replay in the sim).
The big problem is that it couldn't really be replay based as lots of messages will trigger some kind of response from the BlueBus that it expects a response to. At that point, you're stuck re-implementing the protocol in order to simulate the I-Bus.

>Collecting physical components and wiring them up on a bench would be the alternative - but that doesn't seem much easier.
This is what I have always done. Depending on what you want to do, this can be as simple as wiring up a CD53 with a 10k resistor to pull up the bus so the BlueBus and it can speak.

Thanks!
-Ted