tedsalmon/BlueBus #194: Microphone Issues

lennylen, 2024-02-21:
So I finally got an OE microphone from the dealer and i'm running into a few issues:

1. after installing it, my voice didn't sound the best so i decided to make sure it was seated properly in the microphone fork. after playing around with it i was able to click it inside its place thinking that it would solve the clarity. This actually made things worse and now my voice is barely being picked up. 

2. When i'm on a phone call, I can't adjust the call volume at all. I have to use the offset to increase the volume, save it and it seems to stay that way. i'm not sure why this is happening.  

HW: 
BlueBus v2
iPhone 14 Pro - latest iOS
Brand new OE Microphone - 84318380338

FW: 1.3.4

Something very interesting to note about problem #1.. I haven't fully finished testing yet, but it seems that this problem is totally non existent on my Android phone connected to BlueBus. 

I didn't have a chance to test too long but i was able to hear my voice pretty loud and clear which was bizarre to me. I will confirm the split test tomorrow when I have a chance to try it again.

Any ideas on what could be going on for either of these issues?

Thanks

tedsalmon, 2024-02-21:
@lennylen 

1. Make sure the Mic Gain Setting is set to 8dB.
2. 
3. This is on a MID-based car, right? Could I get a log of you making a call and then trying to adjust the call volume? Your aren't the first to report this but I haven't been able to replicate it.
4. 
-Ted

lennylen, 2024-02-21:
Gotcha, 

1. Why 8db specifically for mic gain? I think I did try going through each decibel step to see if it was making a difference or not, but I don't recall hearing much of a difference at 8db. 

1b. any insight on why iphone call quality could be degraded on this setup? -- still needs further testing with android vs ios here. but from the short test i did, android microphone quality to the other end was crystal clear.

2. Yes MID, '01 E53 to be specific. I will report back with a log while making a call. I believe I saw instructions on the wiki on how to capture the log and report it back.

Thanks

closed 2024-04-21 by lennylen