tedsalmon/BlueBus #195: Navigation MKIV does not properly display metadata

FQuimson23, 2024-02-24:
I'm having an issue with the metadata on an MKIV system appearing as it did when I had an MKIII System in previously. The radio in FM also displays correctly but not on Bluetooth. I'm not sure if this will affect anything, but I did swap out my original BM53 radio for a Siemens unit out of an X3 since the radio amplifier seemed to be on its way out.

The vehicle is an E39 Prod Date 01/02 with MKIV upgraded to latest version. I can get the P/N from the radio unit if it's needed. 
[image]
[image]
[image]

FQuimson23, 2024-02-24:
Yeah I might’ve thought it was a little confusing, but that’s the metadata from the radio station on the last picture. Metadata from Bluetooth mode itself is like it is in the first picture, displayed as if I still had the MKIII. 

I included the radio station metadata picture just to show that the function was working on another mode.

tedsalmon, 2024-02-24:
@FQuimson23,

Ah! Damn, I need to source one of the HD Siemens radios -- I didn't know they implemented the static UI for radio metadata! If you access the CLI, you can get the static UI by running this command:

```
set ui 6
```
Here's the relevant code: https://github.com/tedsalmon/BlueBus/blob/master/firmware/application/ui/cli.c#L681

Thanks!
-Ted

tedsalmon, 2024-02-28:
@FQuimson23,

Are we all good here? :)

-Ted

FQuimson23, 2024-02-28:
I haven't been able to test since my E39 decided that it wanted to leak coolant everywhere the day I posted this issue.

Anyways, I'll attach a picture of some P/N and prod. dates that may help you source a Siemens radio with metadata and/or HD Radio capabilities. :) 
[image]

tedsalmon, 2024-03-06:
@FQuimson23

Ah yes, it's just marking its territory :)

Thanks for the screenshot!

-Ted

FQuimson23, 2024-03-30:
Looks like it's all good to go! Finally had some time to get the radiator replaced for the leak and setting the UI to 6 worked. 

[image]

closed 2024-03-30 by FQuimson23