tedsalmon/BlueBus #198: One touch blinkers not working if license plate light broken

mpbecker, 2024-03-19:
Hi Ted, hi all!

I recently noticed an interesting behavior: one of the bulbs of the license plate illumination went out, resulting in the one touch blinker function being broken. Broken means: on one touch, blinkers start blinking in the correct direction, but never stops. To stop it, you have to do a "fixed" (so no one touch) blinking in die other direction.

This has been observed in a 2001 E39 with MID.

tedsalmon, 2024-06-08:
@mpbecker 

Sorry for the lag! Are you sure this was related? It doesn't totally makes sense that one would cause the other. Typically, disabling Parking Lights helps the Comfort Blinkers work consistently...

Thanks!
-Ted

mpbecker, 2024-06-08:
Hi! No worries. Yes, I am pretty shure! I never use Parking Lights, so I did not change any of the configuration. The comfort blinkers stopped behaving normal the moment, the bulb went out and kept this way until I replaced it. Then immediatley worked again. It is an E39 with MID and the "advanced" Cluster, showing you the information that a bulb broke as text (in German something like "Kennzeichenleuchte defekt"). I wondered if this could be some kind of diagnosis message confusing the BlueBus?

tedsalmon, 2024-06-12:
@mpbecker,

What comes to mind is that the bitmap of bulbs engaged may be different when a bulb is out. Would you remind removing a bulb and following this process? https://github.com/tedsalmon/BlueBus/wiki/Gathering-a-Log

I would then just like for you to engage the blinkers in each direction once.

Then, another log with working bulbs would be amazing as well!

In my own E46s, I cannot replicate this problem.

Thanks!
-Ted

mpbecker, 2024-06-12:
Will do! But it will take some time, the E39 lives currently in the garage for some repairs :D

stemo76, 2024-07-19:
I have an e39 and something is wrong with the wiring to the lic plate lights. I also have this issue where the soft touch blinker won't stop.  I need to use the opposite signal and then turn it off.  Its not always the case, seems it depends on the boot sequence.

dferguson08, 2024-07-19:
I've had the same blinker issue surface on my e85. I haven't identified a bulb outage or start-up sequence that's present when the blinker issue occurs, but the issue does appear intermittent.

tedsalmon, 2024-07-23:
If anyone with a broken license plate could grab a log, that would be amazing :)

@dferguson08 -- Are you on Firmware 1.3.4?

tedsalmon, 2026-10-09:
This was resolved in 1.4.20, and I would highly recommend that you all move up to 1.4.43 :)

closed 2026-10-09 by tedsalmon