tedsalmon/BlueBus #203: Not able to autoplay (BC127 BlueBus not able to connect)

HDC67, 2024-06-13:
Ok, firstly this is DIY built BlueBus v1. It has generally worked just fine but seems with the current phone I have, the Autoplay doesn't work. I don't drive the car that much, but I'd probably say it's never worked with this phone. It did work with an older phone. Mostly it works fine but the noticeable failure is the auto play doesn't work. It's not a huge issue hence I only today got around to getting a log.

It seems what happens is:

1) Ignition on
2) It tries to connect to phone
3) Phone seems to think it's connected according to Bluetooth menu
4) BC127 returns "OPEN_ERROR A2DP " including the trailing space.
5) This happens a few times - sometimes the BMBT play/pause toggles whilst this is happening
6) BlueBus gives up
7) I press the play/pause on the BMBT or in the amp and it starts to play properly with no extra changes

Other points of note:
1) If I turn off ignition, it disconnects and the player (PlexAmp) stops. If I say skip the track but leave device paused whilst the car is off, when ignition comes back on it connects properly and autoplay works. Wierd right?
2) I tried both unparing and the BlueBus RESTORE command as well as deleting the device from the phone itself. When I enabled pairing again, the phone popped up immediately saying the car was trying to connect and did I want to allow it (because I'd unpaired on the phone). So clearly the BC127 hadn't completely forgotten about my phone.
3) ERROR: 0x0019 you see when it seems to give up means "Name not found" whatever that means.

Phone is a Oppo Find X5 Pro. It seems it might be a phone issue or at least a phone/BC127 issue. I have 7.3 of the BC127 firmware and I'm using 1.3.4 of BlueBus although I had been using 1.1.18 I think and tried to upgrade hoping it might magically fix it :)

I appreciate that it's not a purchased module but would be interested any ideas, or maybe things to try?

Maybe if someone else has an Oppo Find X5 Pro or other Oppo, does it work properly?

Log looks like this from just before the ignition on, through the trying to connect phase and ultimately the actual playing by pressing play/pause (I think on the BMBT, not the app). This is before I tried to unpair etc.
```
[141850] DEBUG: IBus: RX[8]: 68 06 3B A5 62 01 00 93 [SELF]
[141862] DEBUG: IBus: RX[10]: 68 08 3B A5 62 01 02 7C 7C 9F [SELF]
[141876] DEBUG: IBus: RX[8]: 68 06 3B A5 62 01 00 93 [SELF]
[141888] DEBUG: IBus: RX[17]: 68 0F 3B 21 62 01 40 42 6C 75 65 74 6F 6F 74 68 08 [SELF]
[141903] DEBUG: IBus: RX[8]: 68 06 3B A5 62 01 00 93 [SELF]
[141916] DEBUG: IBus: RX[10]: 68 08 3B A5 62 01 02 7C 7C 9F [SELF]
[141929] DEBUG: IBus: RX[8]: 68 06 3B A5 62 01 00 93 [SELF]
[141941] DEBUG: IBus: RX[17]: 68 0F 3B A5 62 01 06 4E 6F 20 44 65 76 69 63 65 A5 [SELF]
[141957] DEBUG: IBus: RX[10]: 68 08 3B A5 62 01 02 7C 7C 9F [SELF]
[141970] DEBUG: IBus: RX[8]: 68 06 3B A5 62 01 00 93 [SELF]
[141983] DEBUG: IBus: RX[16]: 18 0E 68 39 00 82 00 01 00 01 01 00 01 01 01 C5 [SELF]
[141998] DEBUG: IBus: RX[6]: 3B 04 68 45 00 12 [SELF]
[142009] DEBUG: IBus: RX[8]: A4 06 BF 70 00 00 0A 67 
[142034] DEBUG: IBus: RX[8]: A4 06 BF 70 02 00 0A 65 
[151350] DEBUG: IBus: RX[7]: 44 05 BF 74 04 00 8E 
[159603] DEBUG: IBus: RX[6]: 80 04 BF 11 01 2B 
[159614] DEBUG: Handler: Ignition On
[159623] DEBUG: BT: W: 'BT_STATE ON OFF'
[159633] DEBUG: BT: W: 'LIST'
[159641] DEBUG: BT: W: 'TONE V 0 N C6 L 4'
[159652] DEBUG: BT: W: 'STATUS'
[159661] DEBUG: Handler: Request LCM Redundant Data
[159671] DEBUG: BT: R: 'OK'
[159680] DEBUG: BT: R: 'LIST 74EF4BD03D44 A2DP AVRCP'
[159690] DEBUG: BT: Paired Device 74EF4BD03D44
[159700] DEBUG: BT: W: 'NAME 74EF4BD03D44'
[159711] DEBUG: BT: R: 'LIST 947BE7A7E465 A2DP AVRCP'
[159722] DEBUG: BT: Paired Device 947BE7A7E465
[159731] DEBUG: BT: W: 'NAME 947BE7A7E465'
[159742] DEBUG: BT: R: 'OK'
[159750] DEBUG: BT: R: 'OK'
[159759] DEBUG: BT: R: 'STATE CONNECTED[0] CONNECTABLE[ON] DISCOVERABLE[OFF] BLE[OFF]'
[159772] DEBUG: BT: Got Status CONNECTABLE[ON] DISCOVERABLE[OFF]
[159784] DEBUG: BT: R: 'OK'
[159792] DEBUG: IBus: RX[7]: ED 05 F0 4F 12 11 54 
[159803] DEBUG: BT: R: 'PENDING'
[159811] DEBUG: BT: R: 'PENDING'
[159820] DEBUG: IBus: RX[5]: ED 03 80 14 7A 
[159830] DEBUG: IBus: RX[7]: 00 05 BF 7A 51 61 F0 
[159841] INFO: Handler: Central Locks unlocked
[159851] DEBUG: IBus: RX[6]: 00 04 BF 76 00 CD 
[159862] DEBUG: BT: R: 'NAME 74EF4BD03D44 "Find X5 Pro"'
[159873] DEBUG: Process Name
[159881] DEBUG: Handler: No Device -- Attempt connection
[159892] DEBUG: Connecting to preferred device
[... 275 lines cut ...]
[181607] DEBUG: BT: R: 'AVRCP_PLAY 11'
[181616] DEBUG: BT: Playing
[181675] DEBUG: IBus: RX[17]: 68 0F 3B 21 62 01 40 42 6C 75 65 74 6F 6F 74 68 08 [SELF]
[181690] DEBUG: IBus: RX[8]: 68 06 3B A5 62 01 00 93 [SELF]
[181702] DEBUG: IBus: RX[10]: 68 08 3B A5 62 01 02 3E 20 81 [SELF]
[181716] DEBUG: BT: R: 'AVRCP_MEDIA 11 TITLE: Usual Places'
[181727] DEBUG: BT: R: 'AVRCP_MEDIA 11 ARTIST: Ash'
[181738] DEBUG: BT: R: 'AVRCP_MEDIA 11 ALBUM: Race the Night'
[181749] DEBUG: BT: title=Usual Places,artist=Ash,album=Race the Night
[181762] DEBUG: IBus: RX[8]: 68 06 3B A5 62 01 00 93 [SELF]
[181774] DEBUG: IBus: RX[5]: 3B 03 6A 01 53 
[181784] DEBUG: IBus: RX[7]: 3B 05 68 22 00 00 74 
[181796] DEBUG: IBus: RX[7]: 3B 05 68 22 00 00 74 
[181807] DEBUG: IBus: RX[7]: 3B 05 68 22 00 00 74 
[181979] DEBUG: IBus: RX[8]: 68 06 3B A5 60 01 00 91 [SELF]
[181992] DEBUG: IBus: RX[17]: 68 0F 3B 21 62 01 40 55 73 75 61 6C 20 50 6C 61 7D [SELF]
[182007] DEBUG: IBus: RX[8]: 68 06 3B A5 62 01 00 93 [SELF]
[182020] DEBUG: IBus: RX[20]: 68 12 3B 21 60 00 00 55 73 75 61 6C 20 50 6C 61 63 65 73 56 [SELF]
[182036] DEBUG: IBus: RX[11]: 68 09 3B 21 60 00 01 41 73 68 40 [SELF]
[182051] DEBUG: IBus: RX[29]: 68 1B 3B 21 60 00 02 52 61 63 65 20 74 68 65 20 4E 69 67 68 74 06 06 06 06 06 06 06 1D [SELF]
[182070] DEBUG: IBus: RX[23]: 68 15 3B 21 60 00 04 54 65 6D 70 B0 43 3A 20 41 3A 2B 32 30 2E 30 8A [SELF]
[182087] DEBUG: IBus: RX[8]: 68 06 3B A5 60 01 00 91 [SELF]
[182099] DEBUG: IBus: RX[8]: 68 06 3B A5 60 01 00 91 [SELF]
[182111] DEBUG: IBus: RX[7]: 3B 05 68 22 00 00 74 
```

tedsalmon, 2024-06-21:
HI @HDC67,

I have never even heard of this phone! At a glance, I am seeing that the phone is immediately pausing after the AVRCP play command `MUSIC 11 PLAY` is sent to the handset:

```
[168694] DEBUG: BT: W: 'MUSIC 11 PLAY'
[168721] DEBUG: BT: R: 'OK'
[168729] DEBUG: BT: R: 'AVRCP_PLAY 11'
[168738] DEBUG: BT: Playing
[168796] DEBUG: BT: R: 'AVRCP_PAUSE 11'
[168805] DEBUG: BT: Paused
```

I can't do much about this, sadly. What Music app are you using, and have you tried different ones?

-Ted

HDC67, 2024-06-26:
It's Oppo's flagship phone so whilst it's no Samsung or iPhone it's not what I'd call particularly esoteric :)

Anyway the real problem seems to be before the lines you picked out.

```
[159603] DEBUG: IBus: RX[6]: 80 04 BF 11 01 2B 
[159614] DEBUG: Handler: Ignition On
[159623] DEBUG: BT: W: 'BT_STATE ON OFF'
[159633] DEBUG: BT: W: 'LIST'
[159641] DEBUG: BT: W: 'TONE V 0 N C6 L 4'
[159652] DEBUG: BT: W: 'STATUS'
[159661] DEBUG: Handler: Request LCM Redundant Data
[159671] DEBUG: BT: R: 'OK'
[159680] DEBUG: BT: R: 'LIST 74EF4BD03D44 A2DP AVRCP'
[159690] DEBUG: BT: Paired Device 74EF4BD03D44
[159700] DEBUG: BT: W: 'NAME 74EF4BD03D44'
[159711] DEBUG: BT: R: 'LIST 947BE7A7E465 A2DP AVRCP'
[159722] DEBUG: BT: Paired Device 947BE7A7E465
[159731] DEBUG: BT: W: 'NAME 947BE7A7E465'
[159742] DEBUG: BT: R: 'OK'
[159750] DEBUG: BT: R: 'OK'
[159759] DEBUG: BT: R: 'STATE CONNECTED[0] CONNECTABLE[ON] DISCOVERABLE[OFF] BLE[OFF]'
[159772] DEBUG: BT: Got Status CONNECTABLE[ON] DISCOVERABLE[OFF]
[159784] DEBUG: BT: R: 'OK'
[159803] DEBUG: BT: R: 'PENDING'
[159811] DEBUG: BT: R: 'PENDING'
[159862] DEBUG: BT: R: 'NAME 74EF4BD03D44 "Find X5 Pro"'
[159873] DEBUG: Process Name
[159881] DEBUG: Handler: No Device -- Attempt connection
[159892] DEBUG: Connecting to preferred device
[159902] DEBUG: BT: W: 'OPEN 74EF4BD03D44 A2DP'
[159914] DEBUG: BT: New Pairing Profile
[159923] DEBUG: IBus: RX[7]: 00 05 BF 7D 00 10 D7 
[159934] DEBUG: BT: R: 'PENDING'
[159943] DEBUG: BT: R: 'OPEN_ERROR A2DP '
...
[160387] DEBUG: Handler: Attempting to resolve pairing error
[160398] DEBUG: BT: W: 'OPEN 74EF4BD03D44 A2DP'
[160432] DEBUG: BT: R: 'PENDING'
[160440] DEBUG: BT: R: 'OPEN_ERROR A2DP '
...
[162909] DEBUG: Handler: Attempting to resolve pairing error
[162920] DEBUG: BT: W: 'OPEN 74EF4BD03D44 A2DP'
[162953] DEBUG: BT: R: 'PENDING'
[162962] DEBUG: BT: R: 'OPEN_ERROR A2DP '
[163307] WARNING: BT: Unable to PLAY - AVRCP link unopened
...
[164995] DEBUG: BT: R: 'ERROR 0x0019'
[165432] DEBUG: Handler: Attempting to resolve pairing error
[165443] DEBUG: BT: W: 'OPEN 74EF4BD03D44 A2DP'
[165468] DEBUG: BT: R: 'PENDING'
[166601] DEBUG: BT: R: 'OPEN_OK 10 A2DP 74EF4BD03D44'
[166612] DEBUG: BT: Open A2DP for ID 10
[166623] DEBUG: BT: W: 'VOLUME 10 UP'
[166632] WARNING: BT: Unable to PLAY - AVRCP link unopened
```

tedsalmon, 2025-06-10:
@HDC67 

Sorry for missing this!

```
[168673] DEBUG: BT: R: 'OPEN_OK 11 AVRCP 74EF4BD03D44'
[168683] DEBUG: BT: Open AVRCP for ID 11
[168694] DEBUG: BT: W: 'MUSIC 11 PLAY'
[168721] DEBUG: BT: R: 'OK'
[168729] DEBUG: BT: R: 'AVRCP_PLAY 11'
[168738] DEBUG: BT: Playing
[168796] DEBUG: BT: R: 'AVRCP_PAUSE 11'
[168805] DEBUG: BT: Paused
```

These are all after the examples you pointed out. AVRCP opens, the BlueBus sends a play and immediately receives a "pause".

> The fact it will not completely forget the pairing seems problematic too. I'll delete the pairing and yet it still pops on my phone and tries to connect to the phone after deleting the pairing.
Did you clear Pairings on the BlueBus?

-Ted

HDC67, 2025-06-22:
I cleared pairings on both the BlueBus and the phone multiple times. Yet when I go to repair, it seems as if the BC127 has not actually cleared the pairing since it still tries to connect to my phone (which pops up the message since I deleted the pairing).