tedsalmon/BlueBus #204: E39 blubus pairing

COOPERSPORT1, 2024-06-18:
Not a complete technophobe but not brilliant  either! Picked up my E39 with blubus via CD cables yesterday, previous owner unpaired his iPhone, however my Samsung will not find it as a Bluetooth device.

Left overnight still no joy, unplugged for 20mins still no good.

What's the remedy?

tedsalmon, 2024-06-18:
Did you enable Pairing @COOPERSPORT1 ? :)

COOPERSPORT1, 2024-06-18:
OK actually I take it back, I am a techno biff, how do I do that?

tedsalmon, 2024-06-18:
@COOPERSPORT1,

Do you have Nav or not?

COOPERSPORT1, 2024-06-18:
No nav

COOPERSPORT1, 2024-06-18:
No nav

tedsalmon, 2024-06-18:
@COOPERSPORT1

Is this a US car? If so, you should just have a "PAIR" button on the bottom row of the MID.

COOPERSPORT1, 2024-06-18:
No it's a UK 2003 525d touring. Assuming I need to plug in with my laptop and enable something?

tedsalmon, 2024-06-18:
@COOPERSPORT1,

Can post a picture of your radio, please? If it has a physical "6" button, you can hit that to enable pairing.

-Ted

COOPERSPORT1, 2024-06-18:
Bingo that worked, guess if you don't know you don't know!
Will it pair to 2 phones simultaneously or is that asking a bit much?

COOPERSPORT1, 2024-06-18:
[image]

COOPERSPORT1, 2024-06-18:
Although found and connected it seems to keep dropping the connection

tedsalmon, 2024-06-18:
@COOPERSPORT1

You have an IRIS rather than a MID which means you don't have a UI either. Shoot me an email, please -- support@bluebus.dev

closed 2024-06-18 by tedsalmon