tedsalmon/BlueBus #205: Screen black with e46

AdonisAAA, 2024-06-21:
Hello, 
I've just built a bluebus for myself
And the BootLoader （V1.3 from GitHub） and firmware V1.3 （from Bluebus website） just successfully flashing
But the display screen of the car just turned into a black screen when Bluebus plug in
I just realized that I missed the configuration of the Bluetooth module, May I ask if this is the reason why the bluebus is not working, and if so, I just downloaded the BlueBus_Provisioning_1_3_2 form here, but I'm not sure how to flashing it in to the bluebus

tedsalmon, 2024-06-21:
Check the resistance from I/K-Bus to ground -- sounds like there's a short.

The missing Bluetooth config is not the reason for this, though.

AdonisAAA, 2024-06-23:
It's 80KΩ， looks this resistance is no problem, and the power LED lights up which means at least IBUS can turn on the power

AdonisAAA, 2024-06-23:
And I just run bt_test.bat in BlueBus_Provisioning, it gives me IBUS: ERROR, Does that mean chip TH3122 has some problem？

tedsalmon, 2024-06-23:
Does your TH3122.4 have “1936” written on it?

AdonisAAA, 2024-06-23:
I have 1935 on it

tedsalmon, 2024-06-23:
> I have 1935 on it

Right, I misremembered! Thats part of a bad batch of TH3122.4s — get an ELMOS 10026B instead. Your best bet is to desolder one from the sport steering wheel multi-function control board.

AdonisAAA, 2024-06-23:
Thank you so much, let me try to get a 10026B and try it

tedsalmon, 2024-07-23:
Closing this ticket :)

closed 2024-07-23 by tedsalmon