tedsalmon/BlueBus #206: Audio issues during phone call

archonauto, 2024-07-30:
Hardware 2, Firmware 1.3.4, CD53 in 2003 Z4 (E85) with Carver audio. Mic gain set to 8db.

Having a couple of issues:

1) When call mode is set to "default", no audio plays through the speakers of the car during the call. Had this same issue on firmware 1.3.3, went away when I downgraded to 1.3.0, but returned when I updated to 1.3.4.  If I set call mode "TCU" or "No Mute", the call audio plays, but there are other issues that would prevent me from using them such as no ability to control the volume of the call audio and voice quality is terrible for the other person with many clicks and noise even when the car is parked inside a garage with the engine off. Voice quality is quite clear with no clicking noises for the other person when call mode is set to "default" but then I have no audio.

2) If call volume offset is used (only in "default" call mode), the offset stays applied after the call ends and the music resumes at a louder level. This can be particularly jarring.

archonauto, 2024-07-31:
After a bit of inspection on my car, it’s looking like issue 1 might not actually be a bluebus problem. I’ve been checking the wiring diagrams for the car and following the harness. The bluebus connects to the car’s 54-pin connector, which in turn feeds to the 12 pin phone prep connector in the car’s harness at the bulkhead. Looks like “tel mute”, “tel on”, “tel out +” and “tel out -“ should all run from that connector to the CD53 head unit but “tel mute” is the only one that seems to actually be at the head unit. Looks like I’ll have to run wires for the other 3 (or find the missing ones bundles somewhere in the dash?) and see what happens

tedsalmon, 2024-08-29:
@archonauto,

I think we got most of this solved over email, right? I definitely need to look at issue 3 still, though.

-Ted

archonauto, 2024-08-29:
> @archonauto,
> 
> I think we got most of this solved over email, right? I definitely need to look at issue 3 still, though.
> 
> -Ted

Thanks Ted. I’ve been meaning to follow up with you for a while now so I just replied to the email again. The important bits (issue 1) seem to be sorted out now between the firmware you sent me and a 338 microphone. Issue 2 was still there but doesn’t actually affect me since I don’t run the volume offset high enough to trigger it. So it’s really just issue 3  and the comfort blinker glitches that still remain.

chicodorosario, 2024-12-11:
I have the same issue as mentioned above with issue 1. I installed my newly acquired OEM microphone and doing test calls and it's as mentioned above in default, TCU or even in no mute. 

Please advise

tedsalmon, 2024-12-11:
> I have the same issue as mentioned above with issue 1. I installed my newly acquired OEM microphone and doing test calls and it's as mentioned above in default, TCU or even in no mute.
> 
> Please advise

Can you shoot me an email, please? It's *so much* easier to keep track of issues there! It makes sense that "Default" call mode doesn't work for you though, as I recall you had the DSP amplifier in your Touring.

Thanks!
-Ted

closed 2024-12-15 by tedsalmon