tedsalmon/BlueBus #207: Volume button now replays the song

Water224, 2024-08-10:
I want to first say this product has been amazing and has made my e46 experience much better. I recently installed a e46 bmw sport steering wheel on my base 03 325xi. After the install, I realized that the button on the steering wheel to lower the volume will now cause the song to restart. Im not sure if this is because I installed a different steering wheel so wanted to ask if this has happened/been brought up before? Here is a link to the wheel I purchased if it helps

https://www.ebay.com/itm/186432473653?chn=ps&norover=1&mkevt=1&mkrid=711-117182-37290-0&mkcid=2&mkscid=101&itemid=186432473653&targetid=2299003535955&device=c&mktype=pla&googleloc=9032883&poi=&campaignid=21214315381&mkgroupid=161363866036&rlsatarget=pla-2299003535955&abcId=9407526&merchantid=6296724&gad_source=1&gclid=Cj0KCQjwn9y1BhC2ARIsAG5IY-5TaD9nYwOy0-oqaWQZsxO4okJj7pdIVIu8eiGZ5KitWIcsZA1SlJEaAmZ3EALw_wcB

tedsalmon, 2024-08-29:
@Water224 

Sorry for the lag! At first glance, it sounds like a problem with the MFL. Could you possibly replace the multi-function button circuitry? It's pretty easy to do once you remove the upper trim -- it's about 6 small screws if I recall correctly.

Thanks!
-Ted

tedsalmon, 2025-06-10:
Closing due to age.

closed 2025-06-10 by tedsalmon